cyberbotics/webots #4342: Progress bars difference between themes

ad-daniel, 2022-03-15:
**Describe the Bug**

It seems the progress bars for dark and classic themes differ

[image]
[image]

Both should show the progress percentage.

BenjaminDeleze, 2022-03-21:
I now have the text not at the right place on develop if I use the dark or dusk theme
[image]

ad-daniel, 2022-03-21:
I can reproduce it, but only on develop. I suspect it might be something that is no longer the default in Qt6

closed 2022-03-22 by BenjaminDeleze