cyberbotics/webots #4343: Important performance drop when using medium texture quality

ad-daniel, 2022-03-15:
**Describe the Bug**

Set your OpenGL settings to:
- occlusion: disabled
- texture quality: **high**
- filtering: 4
- shadows: unchecked
- AA: unchecked

run:
`./webots projects/robots/dji/mavic/worlds/mavic_2_pro.wbt`
it takes about 10 seconds to load.

Now change the texture quality from  **high** to **medium**, close webots and run the same command again.
Now it takes ~44 seconds to load.

The effect varies from world to world, but the mavic is one particularly noticeable. What about it is the issue I'm not sure. either way, it doesn't seem a normal behavior when lowering the settings.

omichel, 2022-03-15:
It could be that the down-scaling of the textures is taking time...

ad-daniel, 2022-03-15:
Yes, it stays a long while stuck at 0% in the "finalizing nodes" state, which I assume is where that takes place.
Either way it seems excessive, and considering this world is also in the guided tour, having to wait for 45s for an overall very simple world seems too much. For reference, it takes 34 seconds to load CH_Vens. with the same settings.

ShuffleWire, 2022-03-15:
I've been trying as well, and I could not experience the slowdown, even if I could says that visually the texture is well down-sampled (looking at the characters on the tape of the box, for instance) The world load within a consistent 10sec.
I have a not supported setup, however...
I'm still giving you feedback, hope it could help !

Webots : develop @ 0a240a599de5c8973fbbdf2214d803578ba5d2bd
OS : Debian 11
CPU : Intel(R) Core(TM) i9-8950HK CPU @ 2.90GHz
GPU : None installed (driver not installed)
OpenGL informations :    
System: Linux 5.10.0-12-amd64 x86_64
OpenGL vendor: Intel
OpenGL renderer: Mesa Intel(R) UHD Graphics 630 (CFL GT2)
OpenGL version: 4.6 (Core Profile) Mesa 20.3.5
Available GPU memory: N/A

ad-daniel, 2022-03-15:
Thanks! Did you compile in release mode or debug mode?

ShuffleWire, 2022-03-15:
> Thanks! Did you compile in release mode or debug mode? (I did the test in debug)

Release

ShuffleWire, 2022-03-15:
I've rebuilt in debug, and I could reproduce your issue ! got about 34sec on medium and 12sec on high (counting time in my head, not very accurate xD) But still, that seems reproducible

ad-daniel, 2022-03-15:
Sorry, what I meant to say is that it stays stuck at 0% in the progress bar "finalizing nodes". not that the CPU is at 0% (quite the contrary), but you're right it's quite interesting that the low performs so well if indeed the issue is the down sampling

BenjaminDeleze, 2022-06-09:
It seems that it is indeed the downscale that takes a lot of additional time:
For the mavic.wbt:
on low: 78 images, max 0.07 sec per images, total: between 3-5 seconds of overhead in this function
on medium: 69 images, max 0.3 sec per images, total between 12-15 seconds of overhead.
I will investigate further

PR #4621 (2022-06-10): Enhancement optimize image scaledown
Fix #4343 

Tasks:

- [x] Replace downscale functions by Qt ones
- [x] Add a preference option to select (Qt smooth) or not (Qt Fast) the blur-

Here are the performance test I made (time is in seconds):

In debug mode:

|            | **Qt fast** | **Qt smooth** | **Downscale without Gaussian** | **Downscale with Gaussian** | **Downscale with Gaussian and openmp** |
|------------|-------------|---------------|--------------------------------|-----------------------------|----------------------------------------|
| **Low**    | 0.03        | 0.12          | [... cut]