cyberbotics/webots #4345: Broken robot window on develop

omichel, 2022-03-16:
When opening the e-puck robot window in Firefox on Windows on the develop branch, the layout is completely messed up (see below) and I am getting this error in the console:
```js
The stylesheet http://localhost:1234/robot_windows/e-puck/e-puck.css was not loaded because its MIME type, “text/html”, is not “text/css”.
```
In Chrome, it works fine.
Here is what I get in Firefox:
[image]

BenjaminDeleze, 2022-03-16:
I cannot reproduce it on Ubuntu 20

omichel, 2022-03-16:
I cannot reproduce it any more either. It was probably a problem with my local browser cache.

closed 2022-03-16 by omichel