cyberbotics/webots #4351: robot = Robot() idles due to later/unrelated code

JustusGri, 2022-03-17:
**Describe the Bug**
A line that would seem to have no effect on the PREVIOUS code does.

- I was originally knew something was wrong as I had a constant idle while the code never exited.
- Using print statements "here1" and "here2" I discovered the idle to occur after the initialization of the Robot (something that is imported)
- The only further discovery that I made was that commenting out the call to the function would allow the code to exit
- Then I was able to figure out that the code will exit if I comment out line 187 which is appending to an array

I have no idea how appending to an array affects the robot() call.

**Steps to Reproduce**
1. Go to [My Repo](https://github.com/JustusGri/Webots/blob/main/CSCI3302_lab5.zip)
2. Download the only zip
3. Run the simulation
4. See only a single print statement before line 48 and have idle
5. Comment out line 187 or 263
6. The simulation exits successfully

**Expected behavior**
It is expected to only print "here1" and idle forever if lines 187 or 263 are not commented out.

**Screenshots**
Code in discussion in order:

[image]
[image]
[image]

**System**
 - Operating System:  macOS Monterey 12.1
 - Graphics Card: Radeon Pro 555X 4 GB
 - Processor: 2.2 GHz 6-Core Intel Core i7

**Additional context**
For Intro to Robotics at CU Boulder, Taught By: @alecive
A TA for the class tested the code and the controller exited successfully

ad-daniel, 2022-03-17:
Hi, I think the issue is on the architecture of the controller itself, the reason the messages are not being shown is that you enter an infinite loop from which it never breaks off, hence the timestep never actually advances and therefore is unable to print the information to console.

if at line 151 you replace with the following:
```
k = 0
while len(explore_set) > 0:
    k += 1
    if k > 50:
        break
```

You'll see that it doesn't hang. 

Basically the issue is that the simulation **only** advances if you call `robot.step(timestep)`, which while you're inside the while loop of line 151 never gets called again, hence it keeps looping infinitely, stuck at time 0.

ad-daniel, 2022-03-25:
I'm closing this for now, feel free to re-open it if you have other issues

closed 2022-03-25 by ad-daniel