cyberbotics/webots #4356: Error: wb_supervisor_field_get_count() can only be used with multiple fields (MF).

Justin-Fisher, 2022-03-17:
Webots Supervisor docs say:  
> The wb_supervisor_field_get_count function returns the number of items of a multiple field (MF) passed as an argument to this function. If a single field (SF) or NULL is passed as an argument to this function, it returns -1. Hence, this function can also be used to test if a field is MF (like WB_MF_INT32) or SF (like WB_SF_BOOL).

If you do call `wb_supervisor_field_get_count()` on an SF field, it gives the expected value -1, but also prints a warning: "Error: wb_supervisor_field_get_count() can only be used with multiple fields (MF)."

According to the documentation, returning -1 is the intended useful behavior here, so there should be no need for the warning.  Or, if we aren't supposed to use this on SF fields, then the documentation shouldn't suggest this as legitimate usage.

omichel, 2022-03-18:
I agree, we should either get rid of this warning or remove the fact that this function can be used to test the MF/SF type of a field.

omichel, 2022-03-18:
My preference would go for the removal of the warning, as we don't have a simple way to determine if we have a SF or a MF field type.

Justin-Fisher, 2022-03-18:
My somewhat simple way of testing SF-vs-MF has been checking if 'SF' is a substring of the field's typename, but if this could do it faster, that'd be good.

PR #4361 (2022-03-18): Fixed annoying warning
Fixes #4356.
This function should not display a warning for SF fields as it may be used to determine if a field is SF or MF. See https://cyberbotics.com/doc/reference/supervisor#wb_supervisor_field_get_count