cyberbotics/webots #4363: `wb_supervisor_node_get_field_by_index` not working

Justin-Fisher, 2022-03-18:
I had naively expected that all fields would be gettable by index using `wb_supervisor_node_get_field_by_index`. However this doesn't seem to be the case. E.g., I have a Robot node whose translation field is gettable at index 0, as expected, (and is the same translation field that you get by asking for it by name). However, this Robot also has a `rotation` field, again gettable by name, but this is not among any of the fields gettable by index up to (1 less than) its `node_get_number_of_fields` of 30. Judging by the scene tree, I would expect that this `rotation` field should be the one with index 1, but when I use `wb_supervisor_node_get_field_by_index` to get the field with index 1, it returns an SFFloat field with name FPS and plausible frames-per-second value of 60, but that seems like a very strange field for a Robot node to have! Similarly, from the scene tree, I would expect index 3 to be the Robot's `children` field, but getting the field with index 3 instead gets a `softCFM` field, which again is not one that I would expect a Robot to have!

It looks to me like getting fields by index is not at all working as I'd expect! (Since it seems to get the 0th index right, and seems to return actual fields, rather than gobbedlygook, my guess would be that you're doing pointer-arithmetic with pointers in some other list of fields than the node's list of its own fields.)

Justin-Fisher, 2022-03-21:
Okay, I poked around in `supervisor.c`, and it looks like this is indeed a simple case of switching field `index` and `node->id`.  The following lines in `wb_supervisor_node_get_field_by_index` both provide arguments in the order `index`, `node->id`

> 2305:  [WbFieldRef result = find_field_by_id(index, node->id);](https://github.com/cyberbotics/webots/blob/f5ae03cbfb535553ab7870963a3f3479e913c2a8/src/controller/c/supervisor.c#L2305)
> 2316:  [result = find_field_by_id(index, node->id);](https://github.com/cyberbotics/webots/blob/f5ae03cbfb535553ab7870963a3f3479e913c2a8/src/controller/c/supervisor.c#L2316)

However the definition of `find_field_by_id` instead expects the arguments to be ordered `node_id`, `field_id`.

>188: [static WbFieldStruct *find_field_by_id(int node_id, int field_id) {](https://github.com/cyberbotics/webots/blob/f5ae03cbfb535553ab7870963a3f3479e913c2a8/src/controller/c/supervisor.c#L188)

So it looks like the fix is probably just swapping the argument order in the first two lines linked above?

omichel, 2022-03-21:
Well spotted! Can you please submit a [PR](https://github.com/cyberbotics/webots/edit/master/src/controller/c/supervisor.c#L2305) for patching this?

Justin-Fisher, 2022-03-21:
I'm not great at doing PR's through the Github editor (nor perhaps in any other way), but think I have successfully done so here: [https://github.com/cyberbotics/webots/pull/4366](https://github.com/cyberbotics/webots/pull/4366) 

NOTE:  I'm not set up to re-compile Webots C code, so **I DID NOT TEST THIS!!!**  Someone should compile and test it!

omichel, 2022-03-21:
> NOTE: I'm not set up to re-compile Webots C code, so I DID NOT TEST THIS!!! Someone should compile and test it!

It is pretty easy to compile Webots from the sources by following [these instructions](https://github.com/cyberbotics/webots/wiki). This will save you a lot of time when investigating this kind of problems.

Justin-Fisher, 2022-03-21:
Almost all of my C-coding experience is from the last millennium, so I'm not the right person to take this on.  (And reading this C code reminds me of why I like Python so much better, since C wastes countless lines doing stuff that Python takes care of automatically!)  I did skim through `supervisor.c` looking at all the uses of `field_list` and couldn't find any other place where correcting the order of arguments to `find_field_by_id` should matter.  The code definitely looked wrong before, comparing node id's to field indices and vice versa, and that fits with my observing that it returned mistaken field references that were swapped in exactly this way.  But if fixing that problem causes some other problems (as your note in the PR's thread seems to suggest), then I'm not the right person to try to sort this out...  

(My guess would be that the reason it sort of seemed to work before was that, when you repeatedly got the same field by index, it actually kept not being able to find that field in `field_list` (line 2341) so it kept bloating `field_list` with additional copies, which is obviously suboptimal, but might easily appear to work.  If preventing it from bloating the list in this way is breaking things, then I'd suggest looking closely at lines 2343-2356, which are supposed to generate the new `field_list` entry in cases where the old field couldn't be found, which it usually *couldn't* when it was trying to look it up under swapped id-vs-index.  I don't fully understand this block of code, but maybe it was effectively doing something else that needed to be done too, that now gets skipped when it successfully finds the old field?  You might also wonder whether whatever test it's failing is right, if the test was built around mistaken code?)

Anyway, it'd be nice if you could get it fixed though, because right now `wb_supervisor_get_field_by_index` is very risky to use, since the more nodes you use it on, the greater the risk it'll get its wires crossed and start returning fields from some other node instead.  It's likely also inefficient due to list-bloating, and will get more and more inefficient the more you use it, since lookup time is going to scale up `O(n)` as the list bloats. I'd been hoping to use `get_field_by_index` to systematically introspect the whole scene tree in a controller, but that use is currently impossible, since accessing Node N field F currently prevents later access to Node F Field N, and vice versa.  I guess for now, I'll need to go back to my old approach of looking up fields by name, rather than index...

omichel, 2022-03-22:
> Anyway, it'd be nice if you could get it fixed though

Done.

closed 2022-03-23 by omichel

Justin-Fisher, 2022-03-23:
Hrmmm. I'd been hoping that installing the last nightly build would fix this, but after installing it, Webots won't start, saying "The code execution cannot proceed because libb2-1.dll was not found".  I've tried reinstalling, but that didn't fix it.  Are there additional steps that I'm missing?