cyberbotics/webots #4399: POV and axises values in joystick is not appropriate when using ros

FreshNing, 2022-03-28:
**Describe the Bug**
The pov value type in ros topic is Int8Stamped, so when using ros to get the joystick pov value, I can only read up(1) and down(16) buttons. While using normal cpp API, I can read left(4096),  right(256) buttons and their combinations.
Additionally, the axis value range from -128-127, when at the zero level, the value jumps from 127 to -128, this is not convenient.

**System**
 - Operating System: [Linux Ubuntu 20.04]
 - ROS Version: noetic
 - Joystick: Logitech F710

omichel, 2022-03-28:
We should probably replace `Int8Stamped` with `Int16Stamped` in [RosJoystick.cpp](https://github.com/cyberbotics/webots/edit/master/projects/default/controllers/ros/RosJoystick.cpp). May I ask you to do it and propose your changes in a new Pull Request?

FreshNing, 2022-03-28:
> We should probably replace `Int8Stamped` with `Int16Stamped` in [RosJoystick.cpp](https://github.com/cyberbotics/webots/edit/master/projects/default/controllers/ros/RosJoystick.cpp). May I ask you to do it and propose your changes in a new Pull Request?

I only made some Subtle changes and tested it on ubuntu. Now it works. thanks for replies.

FreshNing, 2022-03-28:
> We should probably replace `Int8Stamped` with `Int16Stamped` in [RosJoystick.cpp](https://github.com/cyberbotics/webots/edit/master/projects/default/controllers/ros/RosJoystick.cpp). May I ask you to do it and propose your changes in a new Pull Request?

I make a new change in [RosJoystick.cpp](https://github.com/cyberbotics/webots/blob/master/projects/default/controllers/ros/RosJoystick.hpp), which fix the situation that joystick button released status can not be detected when using ros. And I have updated the changelog and documentation about joystick.

omichel, 2022-03-28:
Could you please add your changes on #4401?

FreshNing, 2022-03-28:
> Could you please add your changes on #4401?

Changed already.

closed 2022-04-01 by omichel