cyberbotics/webots #4411: JSON generated by HTML5 recording is invalid

Adman, 2022-03-30:
**Describe the Bug**
JSON generated by HTML5 recording is invalid `SyntaxError: Unexpected end of JSON input`

**Steps to Reproduce**
1. Start HTML5 recording
2. Run simulation

**Expected behavior**
JSON generated should be valid

**System**
 - Operating System: Linux Ubuntu 20.04

mrshu, 2022-03-30:
Some more information: we were able to reproduce this with strings which have newlines in them. A short sample:

```
{
 "basicTimeStep":32,
 "ids":"77;90;95;109;154;159;173;218;223;237;282;287;301;346;351;365;410;415;429",
 "labelsIds":"47400;47401;47402;47403;47406;47405",
 "frames":[
{"time":0,"poses":[{"id":77,"translation":"-0.0088 0.0771 0.0208","rotation":"0.7071 0.7071 0 -2.6232"},{"id":90,"translation":"0.3694 0.409 0.0374","rotation":"0 0 -1 1.6027"},{"id":95,"translation":"0 -0.045 -0.0176","rotation":"0 0 1 1.5708"},{"id":109,"translation":"0 0.045 -0.0176","rotation":"0 0 1 1.5708"},{"id":154,"translation":"-0.2277 0.4529 0.0373","rotation":"0 0 1 -1.4684"},{"id":159,"translation":"0 -0.045 -0.0176","rotation":"0 0 1 1.5708"},{"id":173,"translation":"0 0.045 -0.0176","rotation":"0 0 1 1.5708"},{"id":218,"translation":"0.0646 0.4313 0.0373","rotation":"0 0 -1 1.5319"},{"id":223,"translation":"0 -0.045 -0.0176","rotation":"0 0 1 1.5708"},{"id":237,"translation":"0 0.045 -0.0176","rotation":"0 0 1 1.5708"},{"id":282,"translation":"-0.2744 -0.3594 0.0373","rotation":"0 0 1 1.57"},{"id":287,"translation":"0 -0.045 -0.0176","rotation":"0 0 -1 1.5708"},{"id":301,"translation":"0 0.045 -0.0176","rotation":"0 0 -1 1.5708"},{"id":346,"translation":"0.3351 -0.3463 0.0373","rotation":"0 0 1 1.5113"},{"id":351,"translation":"0 -0.045 -0.0176","rotation":"0 0 -1 1.5708"},{"id":365,"translation":"0 0.045 -0.0176","rotation":"0 0 -1 1.5708"},{"id":410,"translation":"-0.0123 -0.3684 0.0373","rotation":"0 0 1 1.5803"},{"id":415,"translation":"0 -0.045 -0.0176","rotation":"0 0 -1 1.5708"},{"id":429,"translation":"0 0.045 -0.0176","rotation":"0 0 -1 1.5708"}]},
{"time":0,"poses":[{"id":77,"translation":"0 0 0"},{"id":90,"rotation":"0 0 -1 1.57","translation":"0.3427 0.2456 0.042"},{"id":154,"rotation":"0 0 -1 1.57","translation":"-0.353 0.3046 0.042"},{"id":218,"rotation":"0 0 -1 1.57","translation":"0 0.1 0.042"},{"id":282,"translation":"-0.323 -0.2647 0.042"},{"id":346,"rotation":"0 0 1 1.57","translation":"0.2863 -0.3552 0.042"},{"id":410,"rotation":"0 0 1 1.57","translation":"-0.0357 -0.2542 0.042"}],"labels":[{"id":47405,"font":"/usr/local/webots/resources/fonts/Tahoma.ttf","rgba":"0,0,255,1","size":0.1,"x":0.81,"y":0.05,"text":"Tester Team"},{"id":47406,"font":"/usr/local/webots/resources/fonts/Tahoma.ttf","rgba":"255,255,0,1","size":0.1,"x":0.05,"y":0.05,"text":"Organizers Testing Team"},{"id":47400,"font":"/usr/local/webots/resources/fonts/Tahoma.ttf","rgba":"0,0,255,1","size":0.1,"x":0.92,"y":0.01,"text":"0"},{"id":47401,"font":"/usr/local/webots/resources/fonts/Tahoma.ttf","rgba":"255,255,0,1","size":0.1,"x":0.05,"y":0.01,"text":"0"}]},
{"time":32,"poses":[{"id":77,"translation":"0 0 0.0042"},{"id":90,"translation":"0.3427 0.245 0.0327"},{"id":154,"translation":"-0.3535 0.3041 0.0326"},{"id":218,"translation":"0 0.0999 0.0321"},{"id":282,"translation":"-0.3235 -0.2624 0.0353"},{"id":410,"translation":"-0.0357 -0.2528 0.0339"}]},
{"time":64,"poses":[{"id":90,"rotation":"0.0381 0.0357 -0.9986 1.5708","translation":"0.3426 0.2438 0.0139"},{"id":154,"rotation":"0.047 0.0073 -0.9989 1.5641","translation":"-0.3545 0.3031 0.0137"},{"id":218,"rotation":"0.0184 0.0154 -0.9997 1.5699","translation":"0 0.0996 0.0122"},{"id":282,"rotation":"-0.0654 0.0481 0.9967 1.5693","translation":"-0.3245 -0.2581 0.0215"},{"id":346,"rotation":"-0.0027 -0.0044 1 1.57","translation":"0.2854 -0.3554 0.042"},{"id":410,"rotation":"-0.0531 0.0519 0.9972 1.5725","translation":"-0.0358 -0.25 0.0176"},{"id":77,"translation":"0 0 -0.0017"}],"labels":[{"id":47402,"font":"/usr/local/webots/resources/fonts/Arial.ttf","rgba":"0,0,0,1","size":0.1,"x":0.45,"y":0.01,"text":"00:09"},{"id":47403,"font":"/usr/local/webots/resources/fonts/Tahoma.ttf","rgba":"255,255,255,1","size":0.05,"x":0.01,"y":0.925,"text":"00:10 - Robot B3 is kicking off.
00:10 - The match (10s) has started."}]},
{"time":96,"poses":[{"id":90,"rotation":"0.1131 0.106 -0.9879 1.5808","translation":"0.3427 0.2463 0.0234"},{"id":154,"rotation":"0.1408 0.0216 -0.9898 1.5586","translation":"-0.3529 0.305 0.0232"},{"id":218,"rotation":"0.055 0.0461 -0.9974 1.5716","translation":"0.0001 0.1009 0.0221"},{"id":282,"rotation":"-0.1935 0.1404 0.971 1.5892","translation":"-0.3238 -0.2614 0.0291"},{"id":346,"rotation":"-0.0083 -0.0139 0.9999 1.5702","translation":"0.2847 -0.3556 0.0319"},{"id":410,"rotation":"-0.1561 0.1528 0.9759 1.5937","translation":"-0.0357 -0.2534 0.026"},{"id":77,"translation":"0 0 0.0028"}]},
```

Note that this is not a valid JSON -- one has to escape the newline for that to be the case:

```
{
 "basicTimeStep":32,
 "ids":"77;90;95;109;154;159;173;218;223;237;282;287;301;346;351;365;410;415;429",
 "labelsIds":"47400;47401;47402;47403;47406;47405",
 "frames":[
{"time":0,"poses":[{"id":77,"translation":"-0.0088 0.0771 0.0208","rotation":"0.7071 0.7071 0 -2.6232"},{"id":90,"translation":"0.3694 0.409 0.0374","rotation":"0 0 -1 1.6027"},{"id":95,"translation":"0 -0.045 -0.0176","rotation":"0 0 1 1.5708"},{"id":109,"translation":"0 0.045 -0.0176","rotation":"0 0 1 1.5708"},{"id":154,"translation":"-0.2277 0.4529 0.0373","rotation":"0 0 1 -1.4684"},{"id":159,"translation":"0 -0.045 -0.0176","rotation":"0 0 1 1.5708"},{"id":173,"translation":"0 0.045 -0.0176","rotation":"0 0 1 1.5708"},{"id":218,"translation":"0.0646 0.4313 0.0373","rotation":"0 0 -1 1.5319"},{"id":223,"translation":"0 -0.045 -0.0176","rotation":"0 0 1 1.5708"},{"id":237,"translation":"0 0.045 -0.0 [... cut ...] "-0.1935 0.1404 0.971 1.5892","translation":"-0.3238 -0.2614 0.0291"},{"id":346,"rotation":"-0.0083 -0.0139 0.9999 1.5702","translation":"0.2847 -0.3556 0.0319"},{"id":410,"rotation":"-0.1561 0.1528 0.9759 1.5937","translation":"-0.0357 -0.2534 0.026"},{"id":77,"translation":"0 0 0.0028"}]},
```

I believe I speak for @Adman as well when I say that we'd be happy to fix this issue, provided you point us to where in the code could one locate the serialization of (presumably JSON) strings.

Thanks!

Adman, 2022-04-13:
@BenjaminDeleze Do you think escaping the label string would solve the issue?

omichel, 2022-04-13:
I believe this line:
https://github.com/cyberbotics/webots/blob/master/src/webots/nodes/utils/WbSupervisorUtilities.cpp#L2185
should be changed to :
```c++
    .arg(labelOverlay->text().replace("\n", "\\n");
```
or the like.

mrshu, 2022-06-08:
Thanks @omichel. I am wondering whether it would make sense to do full-scale JSON escape (on the order of https://stackoverflow.com/questions/7724448/simple-json-string-escape-for-c) and whether the `WbSupervisorUtilities.cpp` are the best place to do that -- if a new export format comes along, it probably might make sense to ensure the escaping could be format-specific at that time.

Do let me know which one would you prefer -- I'll see if I can put together a PR :slightly_smiling_face:

omichel, 2022-06-08:
Yes, a full-scale JSON escape would be ideal, although a simple version might be a first step towards this goal.
And yes, `WbSupervisorUtilities.cpp` is probably the best place to do that for now. We might move it elsewhere later on if we need to use it from another place. I'll be happy to review your PR 😃.