cyberbotics/webots #4420: no color

OZZHaoO, 2022-04-01:
WARNING: Floor (PROTO) > SandyGround (PROTO) > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/sandy_ground/sandy_ground_base_color.jpg: 打开https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/sandy_ground/sandy_ground_base_color.jpg发生错误
WARNING: Floor (PROTO) > SandyGround (PROTO) > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/sandy_ground/sandy_ground_roughness.jpg: 打开https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/sandy_ground/sandy_ground_roughness.jpg发生错误
WARNING: Floor (PROTO) > SandyGround (PROTO) > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/sandy_ground/sandy_ground_normal.jpg: 打开https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/sandy_ground/sandy_ground_normal.jpg发生错误
WARNING: Floor (PROTO) > SandyGround (PROTO) > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/sandy_ground/sandy_ground_occlusion.jpg: 打开https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/sandy_ground/sandy_ground_occlusion.jpg发生错误
WARNING: Road (PROTO) > Pavement (PROTO) > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/pavement/black_stone_pavement_base_color.jpg: 打开https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/pavement/black_stone_pavement_base_color.jpg发生错误
WARNING: Road (PROTO) > Pavement (PROTO) > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/pavement/black_stone_pavement_roughness.jpg: 打开https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/pavement/black_stone_pavement_roughness.jpg发生错误
WARNING: Road (PROTO) > Pavement (PROTO) > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/pavement/black_stone_pavement_normal.jpg: 打开https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/pavement/black_stone_pavement_normal.jpg发生错误
WARNING: Road (PROTO) > Pavement (PROTO) > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/pavement/black_stone_pavement_occlusion.jpg: 打开https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/pavement/black_stone_pavement_occlusion.jpg发生错误
WARNING: Windmill (PROTO) > Shape > PBRAppearance > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/objects/buildings/protos/textures/windmill_base_color.jpg: 打开https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/objects/buildings/protos/textures/windmill_base_color.jpg发生错误
WARNING: Windmill (PROTO) > Shape > PBRAppearance > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/objects/buildings/protos/textures/windmill_roughness.jpg: 打开https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/objects/buildings/protos/textures/windmill_roughness.jpg发生错误
WARNING: Windmill (PROTO) > Shape > PBRAppearance > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/objects/buildings/protos/textures/windmill_metalness.jpg: 打开https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/objects/buildings/protos/textures/windmill_metalness.jpg发生错误
WARNING: Windmill (PROTO) > Shape > PBRAppearance > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/objects/buildings/protos/textures/windmill_normal.jpg: 打开https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/objects/buildings/protos/textures/windmill_normal.jpg发生错误
WARNING: Windmill (PROTO) > Shape > PBRAppearance > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/objects/buildings/protos/textures/windmill_occlusion.jpg: 打开https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/objects/buildings/protos/textures/windmill_occlusion.jpg发生错误
WARNING: SmallManor (PROTO) > Shape > PBRAppearance > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/objects/buildings/protos/textures/small_manor_roof_base_color.jpg: 打开https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/objects/buildings/protos/textures/small_manor_roof_base_color.jpg发生错误
WARNING: SmallManor (PROTO) > Shape > PBRAppearance > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/objects/buildings/protos/textures/small_manor_roof_roughness.jpg: 打开https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/objects/buildings/protos/textures/small_manor_roof_roughness.jpg发生错误
WARNING: SmallManor (PROTO) > Shape > PBRAppearance > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/objects/buildings/protos/textures/small_manor_roof_metalness.jpg: 打开https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/objects/buildings/protos/textures/small_manor_roof_metalness.jpg发生错误
WARNING: SmallManor (PROTO) > Shape > PBRAppearance > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/objects/buildings/protos/textures/small_manor_roof_normal.jpg: 打开https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/objects/buildings/protos/textures/small_manor_roof_normal.jpg发生错误
WARNING: SmallManor (PROTO) > Shape > PBRAppearance > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/objects/buildings/protos/textures/small_manor_roof_occlusion.jpg: 打开https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/objects/buildings/protos/textures/small_manor_roof_occlusion.jpg发生错误
WARNING: SmallManor (PROTO) > Shape > PBRAppearance > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/objects/buildings/protos/textures/small_manor_walls_base_color.jpg: 打开https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/objects/buildings/protos/textures/small_manor_walls_base_color.jpg发生错误
WARNING: SmallManor (PROTO) > Shape > PBRAppearance > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/objects/buildings/protos/textures/small_manor_walls_roughness.jpg: 打开https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/objects/buildings/protos/textures/small_manor_walls_roughness.jpg发生错误
WARNING: SmallManor (PROTO) > Shape > PBRAppearance > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/objects/buildings/protos/textures/small_manor_walls_metalness.jpg: 打开https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/objects/buildings/protos/textures/small_manor_walls_metalness.jpg发生错误
WARNING: SmallManor (PROTO) > Shape > PBRAppearance > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/objects/buildings/protos/textures/small_manor_walls_normal.jpg: 打开https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/objects/buildings/protos/textures/small_manor_walls_normal.jpg发生错误
WARNING: SmallManor (PROTO) > Shape > PBRAppearance > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/objects/buildings/protos/textures/small_manor_walls_occlusion.jpg: 打开https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/objects/buildings/protos/textures/small_manor_walls_occlusion.jpg发生错误
WARNING: SmallManor (PROTO) > Shape > PBRAppearance > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/objects/buildings/protos/textures/small_manor_frame_base_color.jpg: 打开https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/objects/buildings/protos/textures/small_man [... cut ...] ult/worlds/textures/cubic/mountains_back.png: 打开https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/default/worlds/textures/cubic/mountains_back.png发生错误
WARNING: TexturedBackground (PROTO): Cannot download leftIrradianceUrl: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/default/worlds/textures/cubic/mountains_back.hdr: 打开https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/default/worlds/textures/cubic/mountains_back.hdr发生错误

ShuffleWire, 2022-04-01:
Maybe this could help you ? https://github.com/cyberbotics/webots/issues/3574

PS: "Hello", "Please" and a useful description might be appreciated next time ;)

stefaniapedrazzi, 2022-05-31:
I will close this issue.
If it should occur again, please first check the URLs are not blocked by your network.
If it is not the case, then reopen the issue and provide a useful description of the issue with additional details.

closed 2022-05-31 by stefaniapedrazzi

omichel, 2022-10-06:
This problem and solution are described [here](https://answers.ros.org/question/403290/warning-texturedbackground-proto-cannot-download/)