cyberbotics/webots #4424: Webots crashes with large bounding object in streaming mode

ruffson, 2022-04-01:
**Describe the Bug**
In Webots 2022a: When using a world with an elevation map and a bounding object (see example below), webots crashes _if_ it is used in streaming mode with this command:
```
webots --batch --stream="port=5001;mode=x3d;" ./world.wbt 
```
It works fine however running the simulation in normal mode. The problem starts happening for elevation maps at 600 x 600 `xDimension` and `yDimension`. Specifically such large elevation maps which are used as bounding objects, see example code below.

**Steps to Reproduce**
1. Execute the python code below which results in a `world.wbt`
2. Launch webots normally with the world to confirm that it works there: `webots world.wbt`
3. Launch webots in streaming mode: `webots --batch --stream="port=5001;mode=x3d;" ./world.wbt`
4. Observe segmentation fault
5. (Optional) Play around with map dimensions and verify that smaller maps with e.g. 500x500 maps work fine

**Expected behavior**
No crash.

**System**
 - Operating System: Ubuntu 20.04
 - Graphics Card: NVIDIA GeForce RTX 3060

**Additional context**
Python file to create a problematic world:

```python
out_str = [
"""#VRML_SIM R2022a utf8
WorldInfo {
  CFM 0.0001
  ERP 0.4
  basicTimeStep 8
  FPS 32
  contactProperties [
    ContactProperties {
      material1 \"track_material\"
      coulombFriction [
        2
      ]
      forceDependentSlip [
        0.0005
      ]
      softCFM 0.0001
    }
  ]
}
Viewpoint {
  orientation -0.15 0.184 0.97 1.45
  position -0.445 -1.22 1.48
  }
TexturedBackground {
}
TexturedBackgroundLight {
}
DEF GEO Solid {
  translation 0 0 0
  rotation 0 0 1 1.57
  children [
    DEF GEO_FLOOR Shape {
      appearance Soil {
        type \"braun\"
        textureTransform TextureTransform {
          scale 20 20
        }
        IBLStrength 1
      }
      geometry ElevationGrid {
"""]
x_dim = 600
y_dim = 600
heights = [1.] * x_dim * y_dim 

out_str.append("height {}\n".format(str(heights)))
out_str.append("xDimension {}\n".format(x_dim))
out_str.append("yDimension {}\n".format(y_dim))

out_str.append("""
        xSpacing 0.05
        ySpacing 0.05
      }
    }
  ]
  boundingObject USE GEO_FLOOR
}
""")

with open("world.wbt", "w") as text_file:
    text_file.write("".join(out_str))
```

BenjaminDeleze, 2022-04-04:
I can reproduce it, I will investigate.

ShuffleWire, 2022-04-04:
Is there any chance it could help fixing this issue ? https://github.com/cyberbotics/webots/issues/3917 I'm putting it there, just in case :)

BenjaminDeleze, 2022-04-05:
I don't think so, the error will normally be fixed in #4426

BenjaminDeleze, 2022-04-05:
The fix is now merged and will be integrated in the next nightly builds

PR #4426 (2022-04-05): Fix crash export elevation grid
Fix #4424 

The arrays were previously allocated on the stack.
For very large elevation grids, it could overload the stack.
So now the arrays are allocated on the heap where there is more room
closed 2022-04-05 by BenjaminDeleze