cyberbotics/webots #4431: Crashed when loading a world created by 2020b.

zapplelove, 2022-04-05:
Here is my robot world created by webots version 2020b:
[world.zip](https://github.com/cyberbotics/webots/files/8416336/world.zip)
and when I want to load it using the latest version, app crashed.

ad-daniel, 2022-04-06:
Since R2022a  there have been significant changes to the orientation of robots/worlds, so in order be able to open it with this version requires adapting your world, more details are available here: https://github.com/cyberbotics/webots/wiki/How-to-adapt-your-world-or-PROTO-to-Webots-R2022a

When loading a world, Webots reads the headers of protos and worlds (i.e something like `#VRML_SIM R2020b utf8`) and if it has been created with a version prior to R2022a it will attempt to do some of the necessary adjustments itself (not everything). There's no guarantee everything will work afterwards but it shouldn't crash either so I will investigate it. However in your case it might be faster to create a new world in R2022a, the robot appears to be in the correct orientation already (don't forget to change the header in the proto however), meaning you probably only need to adapt the conveyor belt (of which you could use the one provided in Webots `ConveyorBelt`)

zapplelove, 2022-04-06:
Thanks a lot. I have changed the `belt.proto` using the one provided in Webots ConveyorBelt, and it worked. 

> By the way, because I am Chinese, I can't get `resources` online when using webots, I have to compile it. So is it possible to use local resources instead of online？

closed 2022-04-06 by zapplelove