cyberbotics/webots #4441: Ros controller crashes when requesting node information from device with a space in the name

nathanieltagg, 2022-04-07:
**Describe the Bug**
Querying the ROS controller for /RobotName/supervisor/get_from_device crashes the ROS controller if the device name has a space in it.

**Steps to Reproduce**
1. Load a simulation with the MiR robot and the default ROS controller
2. `rosservice call /Mir100/imu_data_accelerometer/get_model 1`
The controller crashes.

**Additional context**
The problem is obviously that the ROS controller advertises devices using the Ros::fixedNameString to remove ros-unfriendly characters.  This is handled fine when making /<controller>/<device>/... calls, but the /supervisor simply uses the un-translated name.

This can be fixed in several ways:
- Disallow spaces in device names when using the ros controller (e.g. generate a warning or error message in Ros.cpp when setting up the device table)
- Have the RosSupervisor class run through all available devices, checking for a match using the fixedNameString
- Change the call so that RosDevice handles this kind of request instead of the Supervisor.
 
In addition, RosSupervisor.cpp should include a check for a null device pointer around line 469 to avoid crashes.

Thanks,
Nathaniel

ad-daniel, 2022-04-11:
Isn't it the other way around? It crashes if you `get_from_device` while providing as argument the advertised (i.e. cleaned) name whereas it works when passing the actual name. For instance when calling
```
rosservice call /Mir100/supervisor/get_from_device "imu_data compass"
```
it checks the string `"imu_data compass"` against the `mDeviceList` list [here](https://github.com/cyberbotics/webots/blob/734324b084d8e4b52b37e8de0fd82beee1344f21/projects/default/controllers/ros/Ros.cpp#L234), which are instances of Device objects (technically RosDevice which wraps Device) and [results](https://github.com/cyberbotics/webots/blob/734324b084d8e4b52b37e8de0fd82beee1344f21/src/controller/cpp/Device.cpp#L25-L27) in a call of `wb_robot_get_device` which will find `"imu_data compass"` just fine.

The issue appears to be the fact that if we provide the sanitized name (which is also what we [get](https://github.com/cyberbotics/webots/blob/734324b084d8e4b52b37e8de0fd82beee1344f21/projects/default/controllers/ros/Ros.cpp#L450) when calling `/robot/get_device_list`) we do indeed get a crash of the controller since the device with such name does not exist

nathanieltagg, 2022-04-11:
Yes, you are correct - using the unsanitized name will work correctly when accessing the supervisor/ functions. I didn't check using the unsantized name against the device functions.  It would be fine if the `get_from_device` gave the unsanitized name instead, but that calls into question why the name is sanitized at all.

Anyway, the API is inconsistent: I cannot request a list of devices and then query that list without a crash.

closed 2022-04-13 by ad-daniel