cyberbotics/webots #4442: Bug during sharing to webots.cloud

ThomasOliverKimble, 2022-04-07:
**Describe the Bug**
Uploading to webots.cloud from webots not working

**Steps to Reproduce**
1. Open Webots (built from develop branch)
2. Click in the share button
3. Click on either `Upload your scene` or `Record and upload animation` and we get the following error message: 
[image]

**Expected behavior**
The scene or animation should be uploaded to webots.cloud

**System**
 - Operating System: Linux Ubuntu 20.04
 - Graphics Card: NVIDIA GeForce GTX 1060 6GB

omichel, 2022-04-12:
@ThomasOliverKimble: Did you fix it in https://github.com/cyberbotics/webots-cloud/pull/7? If so, can we close this issue?

ThomasOliverKimble, 2022-04-12:
> @ThomasOliverKimble: Did you fix it in [cyberbotics/webots-cloud#7](https://github.com/cyberbotics/webots-cloud/pull/7)? If so, can we close this issue?

Still not working, I will look into this today.

ThomasOliverKimble, 2022-05-25:
Yes, was fixed with #4544.

closed 2022-05-25 by ThomasOliverKimble