cyberbotics/webots #4444: Unable to undo delete, dangerous!

pluveto, 2022-04-10:
**Describe the Bug**
I'm working hard on my project, but I unintentionally select a group of objects instead of that one I want to delete.

I press delete, everything is gone.

I press undo, not working. Very sad.

**Steps to Reproduce**
1. Create new project with initial floor model
2. Click on floor, press delete
3. Whole scene is deleted
4. Ctrl+Z undo not working

**Expected behavior**
Should undo.

**Screenshots**
[image]

**System**
 - Operating System: Microsoft Windows 10.0.19043.1586
 - Graphics Card: GTX 780TI

ad-daniel, 2022-04-11:
> but I unintentionally select a group of objects instead of that one I want to delete. I press delete, everything is gone.

I'm not sure to understand, do you mean you select multiple object and press delete or you select a node of type Group then delete it?
In the first case, I don't even believe it's possible to select multiple objects (although it would be nice). In the second case, then it's normal that deleting a group will delete all the objects it contains.

> Steps to Reproduce

I can't reproduce this behavior, deleting the Floor only deletes this object for me, not everything.
Also your screenshot looks entirely black to me. Can you provide more details?

ShuffleWire, 2022-04-11:
I'm taking the opportunity to add that also resetting (not reloading) a world on which I just added nodes without saving end up deleting them... Maybe some common issues there ?

pluveto, 2022-04-12:
They were all deleted because I clicked the floor but the whole scene was selected
And the delete operation is indeed irreversible

closed 2022-04-13 by pluveto