cyberbotics/webots #4450: Crash when selecting field in scene tree

ad-daniel, 2022-04-13:
**Describe the Bug**
Webots crashes when attempting to select a url field.

**Steps to Reproduce**
1. Open the following world
```
#VRML_SIM R2022a utf8
WorldInfo {
}
Viewpoint {
}
Background {
  skyColor [
    0.15 0.45 1
  ]
}
Transform {
  children [
    Shape {
      geometry Mesh {
      }
    }
  ]
}
Transform {
  children [
    Shape {
      geometry Mesh {
      }
    }
  ]
}

```
3. Expand the first transform (only this one), and add a random mesh to the Mesh node 
4. Expand the second Transform, when you attempt to select the `url` field Webots crashes.

ad-daniel, 2022-04-13:
It appears to be a rather nasty bug, doing the process in the opposite order does not crash webots. Basically when loading the world, both meshes are initialized  and are empty
```
mesh  node pointer     => number of triangles
M1    (0x7f098000d990) => 0
M2    (0x7f098000ccb0) => 0
```
After populating the first node
```
mesh  node pointer     => number of triangles
M1    (0x7f098000d990) => 968
M2    (0x7f098000ccb0) => 21878
```

So something, somewhere is overflowing.

BenjaminDeleze, 2022-04-13:
I could reproduce the bug, but not entirely like you.

- If I have no backgroud => no crash
- With a background => crash but when selecting the second `shape`
- Additional problem (maybe not related with this issue) when removing the url from the mesh node, it does not remove the mesh from the 3D scene

ad-daniel, 2022-04-13:
> With a background => crash but when selecting the second shape
Which field crashes it appears to depend on how you expand the menus. If you expand them by double-clicking the field, it crashes on the Shape field. If you expanding using the arrow, it crashes on the url field (or castShadows for that matter)

> Additional problem (maybe not related with this issue) when removing the url from the mesh node, it does not remove the mesh from the 3D scene

In theory this was added in develop, or at least I know I did so when implementing the cache but you're right, it doesn't appear to anymore. I'll look into it

PR #4795 (2022-07-01): Fix crash in triangle mesh cache
Fixes #4450.

The life cycle of the `WbTriangleMesh` object wasn't correct: it is created and deleted in `WbTriangleMeshGeometry`, but also deleted in `WbTriangleMeshCache` without updating the pointer in `WbTriangleMeshGeometry`. Hence after deleting the `WbTriangleMesh` in `WbTriangleMeshCache`, `WbTriangleMeshGeometry.mTriangleMesh` was pointing to some memory that was released. Therefore the `WbTriangleMesh` pointed to by `WbTriangleMeshGeometry` was containing corrupted data.

I believe there is no need to delete this `WbTriangleMesh` object in `WbTriangle [... cut]
closed 2022-07-01 by ad-daniel