cyberbotics/webots #4483: Web component broken in develop

ad-daniel, 2022-04-27:
**Describe the Bug**
As emerged here https://github.com/cyberbotics/webots/pull/4192#pullrequestreview-887300963, for the Scara robot (among others) the web component doesn't appear to be placing the sphere denoting the joint anchor at the correct spot, hence it ends up looking as if the point of rotation is incorrect. 

Since then, it seems that other issues have emerged:

develop branch:
- some textures no longer show, for instance: https://cyberbotics.com/doc/guide/bb8?version=develop
- boundingObjects and joint representation no longer showing https://cyberbotics.com/doc/guide/scara-epson-t6?version=develop

released branch, on firefox:
- for example: https://cyberbotics.com/doc/guide/khepera4
`Cross-Origin Request Blocked: The Same Origin Policy disallows reading the remote resource at https://googleads.g.doubleclick.net/pagead/id. (Reason: CORS request did not succeed). Status code: (null).`

BenjaminDeleze, 2022-04-28:
> * some textures no longer show, for instance: https://cyberbotics.com/doc/guide/bb8?version=develop
> * boundingObjects and joint representation no longer showing https://cyberbotics.com/doc/guide/scara-epson-t6?version=develop

This is certainly, at least for the bounding objects, because the `.x3d` were not regenerated since the last changes in WebotsJS
I will handle that.

> * for example: https://cyberbotics.com/doc/guide/khepera4

I did not manage to get an error (either Firefox or Chrome)

ad-daniel, 2022-04-28:
> I did not manage to get an error (either Firefox or Chrome)

We've looked into it yesterday and it doesn't appear to be related to the web component but rather to the google search bar at the very bottom of some pages, which I (we) didn't even know existed

PR #4484 (2022-04-29): Fix web component
- [x] Fix bounding objects
- [x] Fix missing textures
- [x] Modernize the viewer to use WebotsView
  - [x] Locally
  - [x] On the server
- [x] Fix joint representation (small sphere)

fix #4483 

You can check it here: https://cyberbotics.com/doc/guide/youbot?version=fix-web-component
closed 2022-05-06 by BenjaminDeleze