cyberbotics/webots #4520: Feature Request: Supervisor method to get image object of Viewpoint rendering

jgueldenstein, 2022-05-09:
When using webots for reinforcement learning using the OpenAI gym environment, the `render()` method needs to be implemented quite hacky. In reinforcement learning it is often required to have qualitative results (video renderings) to get insight on the learning process. 

One option is to attach a camera node to the robot and move it to the Viewpoint, then use the `getImage()` method of the Camera node.

Another option is to use the `exportImage()` method of the Supervisor. This saves the image to the filesystem though, and it has to be loaded again. This is obviously slow.

A better option would be to have a method of the Supervisor such as `getViewpointImage()` which would return the image similar to the `getImage()` method of the Camera node

ad-daniel, 2022-05-09:
Thank you for the suggestion, it is unclear to us what specifically is problematic on your end about the current implementation. Is it the performance or the hassle of having to add the camera node?
 
Additionally:
- exportImage: loading from file is slow yes, but not significantly slower than calling getImage with Python, since calling `getImage` (contrary to the C counterpart) requires copying the image from GPU to CPU under the hood rather than accessing it directly, which is fairly slow as well. What might add additional overhead is the compression to jpeg when it is exported, in which case adding the possibility of exporting a raw image might be sufficient to achieve comparable performance to a getImage call. What indeed might be an issue with this method is the fact that in the presence of camera overlays, these would be visible as well in the acquisition, which might not be what you want
- even with a `getViewpointImage` function, you'd still need to manually move the viewpoint prior to any acquisition, which doesn't appear significantly different from the current approach that involves having to move the supervisor that contains the camera node. So in the end, it seems the only difference would be the precisely the fact that you currently need this intermediary step of adding a camera node, or am I missing something?

SammyRamone, 2022-05-13:
I tried using the `exportImage()` method to get images from the simulation and encountered two issues:
1. If the rendering is deactivated in the simulation (which it typically is in a RL training process), the returned image is only a black window with "No Rendering" (the same that is shown in the GUI if rendering is deactivated). I did not find any method to activate the rendering. The `supervisor.simulationSetMode(mode)` can be used to set to `SIMULATION_MODE_REAL_TIME` speed but this does not enable rendering if it was previously disabled. Therefore one would need to constantly render the simulation to get some images at some point.
2. Position of the rendered images can be influenced by modifying the viewpoint, but the resolution depends on the size of the Webots window. This is also very unpractical as it is not really possible to chose the resolution.

Therefore I don't think, that the `exportImage()` function is usable in this case.
This only leaves the option of adding an extra camera to the robot. This is, in my opinion, not the best approach as each robot model needs to be modified.

I also searched for other code bases that use Webots as a basis for OpenAI Gym environments. All of them had no render option, which is probably due the missing option of getting the image from the code. I think this is unfortunate, as this is a standard use case in RL.

omichel, 2022-05-13:
I believe the best option is to have a supervisor robot equipped with a camera that will take pictures of the scene (you can choose the camera position, orientation, resolution, etc.) and save them as a raw RGB file which the robot controller programs could read without too much overhead (no compression / decompression). If you need extra performance, you can save the image in a memory mapped file.

jgueldenstein, 2022-05-16:
The problem with this solution is, that only one robot controller can be executed per process. This would mean that inter-process communication is required to do this solution.

Since it is a quite common requirement in RL, I believe a simple solution to enable the rendering and get an image from the viewpoint is quite beneficial to many.

omichel, 2022-05-16:
> The problem with this solution is, that only one robot controller can be executed per process. This would mean that inter-process communication is required to do this solution.

Yes, but if the supervisor controller shares images in memory mapped files, it is super easy for robot controllers to read these images. I recently developed a [simple example of memory mapped files](https://github.com/omichel/mmap) which you can re-used for this purpose.

Note: we are currently [working on a new system allowing to isolate Webots and robot controllers in different docker containers](https://github.com/cyberbotics/webots/pull/4344). The new system required to use memory mapped files for images and once complete, it would be super-easy for any robot controller to access the camera image of an other robot controller, thus providing a zero-copy performance to sharing camera images across Webots controllers.

SammyRamone, 2022-05-19:
Since implementing a custom IPC solution seemed to be more work, I used the earlier mentioned approach of adding an additional camera to the robot, which was successful. In the following I will leave a few pointers on how to implement this if somebody else stumbles on this issue in the future:

1. Add a field to your robot model. See [here](https://github.com/bit-bots/wolfgang_robot/blob/f43a6c1459266293e20c3b2d925e1e107472473f/wolfgang_webots_sim/protos/Wolfgang.proto#L41) and [here](https://github.com/bit-bots/wolfgang_robot/blob/f43a6c1459266293e20c3b2d925e1e107472473f/wolfgang_webots_sim/protos/Wolfgang.proto#L70-L73)
2. Fill the field in the world file. See [here](https://github.com/bit-bots/wolfgang_robot/blob/f43a6c1459266293e20c3b2d925e1e107472473f/wolfgang_webots_sim/worlds/deep_quintic_wolfgang.wbt#L48-L57)
3. Get the camera image and convert it to the used RGB array format of OpenAI Gym. See [here](https://github.com/bit-bots/deep_quintic/blob/da67d87f5f3cab91bb4b00e676f7a38682f3ab72/deep_quintic/simulation.py#L439-L451)

In my solution, the camera just moves with the robots base. This could be improved by adding some transform that is adapted at each step, but I did not spend time implementing it since I only use this for debug purposes where it does not matter.