cyberbotics/webots #4595: webots build is broken on mingw64 when python_home is not set

Kreijstal, 2022-06-03:
**Describe the Bug**
A clear and concise description of what the bug is.

I am running this on mingw, I simply clone the repo, install packages, and do make, this is the re-attempt, after already compiling almost everything.
```
$ make
#
# * dependencies *
#
#
# * ode *
#
#
# * glad *
#
#
# * wren *
#
#
# * webots (core) *
#
#
# * controller library *
#
#
# C controller library (release)
#
# 'VISUAL_STUDIO_PATH' not set, skipping Controller.lib
# 'VISUAL_STUDIO_PATH' not set, skipping Controller.lib (32 bit)
#
# C++ controller library (release)
#
#
# Java controller library (release)
#
# Java not installed or 'JAVA_HOME' not set, skipping Java API
# Matlab controller library (release)
#
# generating matlab
#
# Python 3.7 controller library (release)
#
# Python 3.7 not installed or 'PYTHON37_HOME' not set, skipping Python 3.7 API
#
# Python 3.8 controller library (release)
#
# Python 3.8 not installed or 'PYTHON38_HOME' not set, skipping Python 3.8 API
#
# Python 3.9 controller library (release)
#
# Python 3.9 not installed or 'PYTHON39_HOME' not set, skipping Python 3.9 API
#
# Python 3.10 controller library (release)
#
# Python 3.10 not installed or 'PYTHON310_HOME' not set, skipping Python 3.10 API
#
# * resources *
#
# *** translations ***
#
# *** projects ***
# make release plugins/physics
# make release libraries/generic_robot_window
# copying generic_robot_window.dll
# make release libraries/qt_utils
# make release plugins/robot_windows/generic
# make release plugins/robot_windows/generic_window
# make release controllers/void
#
# * projects *
#
# !!! languages !!!
#
# *** controllers ***
# make release driver
# make release slave
# make_java release Driver
# Java not installed or 'JAVA_HOME' not set, skipping Java API
# make_java release Slave
# Java not installed or 'JAVA_HOME' not set, skipping Java API
# make release e-puck_position_supervisor
#
# !!! default !!!
# make release controllers/braitenberg
# make release libraries/vehicle
# make release c/car
# copying car.dll
'VISUAL_STUDIO_PATH' environmental variable not set or Microsoft Visual Studio not installed, skipping car.lib
# make release c/driver
# linking build/release/driver.o driver.dll
# copying driver.dll
'VISUAL_STUDIO_PATH' environmental variable not set or Microsoft Visual Studio not installed, skipping driver.lib
# make release cpp/driver
# linking build/release/Driver.o CppDriver.dll
# copying CppDriver.dll
# make release cpp/car
# linking build/release/Car.o CppCar.dll
# copying CppCar.dll
# make release java
# Java not installed or 'JAVA_HOME' not set, skipping Java API
# make python python3.10
Unable to open file /home/kreij/webots/lib/controller/python39/vehicle.py: No such file or directory
make[4]: *** [Makefile:98: /home/kreij/webots/lib/controller/python39/vehicle.py] Error 1
make[3]: *** [Makefile:43: python] Error 2
make[2]: *** [Makefile:37: libraries/vehicle.Makefile] Error 2
make[1]: *** [Makefile:23: release] Error 2
make: *** [Makefile:120: webots_target] Error 2
```

**Expected behavior**
If python is not found, then it shouldn't attempt to open "vehicle.py"?

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System**
 - Windows 10

Python is actually installed, it just don't have PYTHON39_HOME set, because I have no idea where is that supposed to be in mingw

omichel, 2022-06-03:
I believe you should install at least one version of Python 3 from https://python.org, see https://github.com/cyberbotics/webots/wiki/Windows-optional-dependencies/#python

Kreijstal, 2022-06-03:
@omichel well, I installed python3.10, set the env variable, and it still thinks about python 3.9
Unable to open file /C/Users/kreij/scoop/apps/msys2/2022-01-28/home/kreij/webots/lib/controller/python39/vehicle.py: No such file or directory
make[4]: *** [Makefile:98: /C/Users/kreij/scoop/apps/msys2/2022-01-28/home/kreij/webots/lib/controller/python39/vehicle.py] Error 1
make[3]: *** [Makefile:45: python] Error 2
make[2]: *** [Makefile:37: libraries/vehicle.Makefile] Error 2
make[1]: *** [Makefile:23: release] Error 2
make: *** [Makefile:120: webots_target] Error 2

I think I will have to install python 3.9 as well, but then it's not at least 1 version

omichel, 2022-06-03:
Hum... Probably something should be fixed here. Feel free to propose a patch if you find the problem.