cyberbotics/webots #4598: Segmentation fault on my mingw build

Kreijstal, 2022-06-03:
**Describe the Bug**
Hmm when finishing making and trying running webots, it loads fine, everything works perfectly until the last moment at the end of "finalizing nodes" at 99%

**Steps to Reproduce**
I type "webots" after compiling

**System**
 - Windows 10 x6

**Additional context**
strace inside mingw revealed the following
<details><summary>Strace output</summary>
<p>

```
--- Process 13396 created
--- Process 13396 loaded C:\Windows\System32\ntdll.dll at 00007ff90bd10000
--- Process 13396 loaded C:\Windows\System32\kernel32.dll at 00007ff90b7e0000
--- Process 13396 loaded C:\Windows\System32\KernelBase.dll at 00007ff9098e0000
--- Process 13396 loaded C:\Windows\System32\msvcrt.dll at 00007ff90b670000
--- Process 13396 thread 20168 created
--- Process 13396 loaded C:\Windows\System32\shlwapi.dll at 00007ff90b1d0000
--- Process 13396 loaded C:\Windows\System32\sechost.dll at 00007ff90ba70000
--- Process 13396 loaded C:\Windows\System32\rpcrt4.dll at 00007ff90b460000
--- Process 13396 thread 19168 created
--- Process 23444 created
--- Process 23444 loaded C:\Windows\System32\ntdll.dll at 00007ff90bd10000
--- Process 23444 loaded C:\Windows\System32\kernel32.dll at 00007ff90b7e0000
--- Process 23444 loaded C:\Windows\System32\KernelBase.dll at 00007ff9098e0000
--- Process 23444 loaded C:\Windows\System32\advapi32.dll at 00007ff909e90000
--- Process 23444 loaded C:\Windows\System32\msvcrt.dll at 00007ff90b670000
--- Process 23444 thread 23356 created
--- Process 23444 loaded C:\Windows\System32\sechost.dll at 00007ff90ba70000
--- Process 23444 loaded C:\Windows\System32\rpcrt4.dll at 00007ff90b460000
--- Process 23444 thread 3560 created
--- Process 23444 thread 24252 created
--- Process 23444 loaded C:\Windows\System32\ole32.dll at 00007ff90bba0000
--- Process 23444 loaded C:\Windows\System32\ucrtbase.dll at 00007ff909bb0000
--- Process 23444 loaded C:\Windows\System32\d3d9.dll at 00007ff8b3920000
--- Process 23444 loaded C:\Windows\System32\gdi32.dll at 00007ff90bb70000
--- Process 23444 loaded C:\Windows\System32\combase.dll at 00007ff90a3c0000
--- Process 23444 loaded C:\Windows\System32\win32u.dll at 00007ff909d00000
--- Process 23444 loaded C:\Windows\System32\gdi32full.dll at 00007ff909420000
--- Process 23444 loaded C:\Windows\System32\user32.dll at 00007ff90b2c0000
--- Process 23444 loaded C:\Users\kreij\scoop\apps\msys2\2022-01-28\mingw64\bin\libassimp-5.dll at 00007ff863f50000
--- Process 23444 loaded C:\Windows\System32\msvcp_win.dll at 00007ff9097d0000
--- Process 23444 loaded C:\Users\kreij\scoop\apps\msys2\2022-01-28\mingw64\bin\libgcc_s_seh-1.dll at 00007ff8e0d20000
--- Process 23444 loaded C:\Windows\System32\glu32.dll at 00007ff8df2b0000
--- Process 23444 loaded C:\Windows\System32\IPHLPAPI.DLL at 00007ff908860000
--- Process 23444 loaded C:\Users\kreij\scoop\apps\msys2\2022-01-28\mingw64\bin\libminizip-1.dll at 00007ff8e0bc0000
--- Process 23444 loaded C:\Windows\System32\opengl32.dll at 00007ff8df2e0000
--- Process 23444 loaded C:\Users\kreij\scoop\apps\msys2\2022-01-28\mingw64\bin\libopenal-1.dll at 00007ff8880f0000
--- Process 23444 loaded C:\Users\kreij\scoop\apps\msys2\2022-01-28\mingw64\bin\libstdc++-6.dll at 00007ff888350000
--- Process 23444 loaded C:\Windows\System32\shell32.dll at 00007ff90aa00000
--- Process 23444 loaded C:\Users\kreij\scoop\apps\msys2\2022-01-28\home\kreij\webots\msys64\mingw64\bin\libpico.dll at 0000000064340000
--- Process 23444 loaded C:\Users\kreij\scoop\apps\msys2\2022-01-28\home\kreij\webots\msys64\mingw64\bin\lua52.dll at 00007ff888fa0000
--- Process 23444 loaded C:\Users\kreij\scoop\apps\msys2\2022-01-28\home\kreij\webots\msys64\mingw64\bin\ode.dll at 00007ff865f90000
--- Process 23444 loaded C:\Users\kreij\scoop\apps\msys2\2022-01-28\home\kreij\webots\msys64\mingw64\bin\zlib1.dll at 00007ff8e0b90000
--- Process 23444 loaded C:\Users\kreij\scoop\apps\msys2\2022-01-28\home\kreij\webots\msys64\mingw64\bin\OIS.dll at 000000006e3c0000
--- Process 23444 loaded C:\Users\kreij\scoop\apps\msys2\2022-01-28\home\kreij\webots\msys64\mingw64\bin\openvr_api.dll at 00007ff888e50000
--- Process 23444 loaded C:\Windows\System32\oleaut32.dll at 00007ff909dc0000
--- Process 23444 loaded C:\Users\kreij\scoop\apps\msys2\2022-01-28\mingw64\bin\libfreetype-6.dll at 00007ff888d90000
--- Process 23444 loaded C:\Users\kreij\scoop\apps\msys2\2022-01-28\mingw64\bin\Qt5Network.dll at 00007ff864e50000
--- Process 23444 loaded C:\Users\kreij\scoop\apps\msys2\2022-01-28\mingw64\bin\Qt5Core.dll at 00007ff863a00000
--- Process 23444 loaded C:\Users\kreij\scoop\apps\msys2\2022-01-28\mingw64\bin\Qt5Gui.dll at 00007ff863360000
--- Process 23444 loaded C:\Windows\System32\crypt32.dll at 00007ff9095c0000
--- Process 23444 loaded C:\Windows\System32\ws2_32.dll at 00007ff90b740000
--- Process 23444 loaded C:\Windows\System32\ws2_32.dll at 000001e50c790000
--- Process 23444 unloaded DLL at 000001e50c790000
--- Process 23444 loaded C:\Users\kreij\scoop\apps\msys2\2022-01-28\mingw64\bin\Qt5PrintSupport.dll at 00007ff888750000
[... 533 lines cut ...]
--- Process 23444 thread 20696 exited with status 0xc0000005
--- Process 23444 thread 1904 exited with status 0xc0000005
--- Process 23444 thread 11804 exited with status 0xc0000005
--- Process 23444 thread 10444 exited with status 0xc0000005
--- Process 23444 thread 21592 exited with status 0xc0000005
--- Process 23444 thread 23200 exited with status 0xc0000005
--- Process 23444 thread 5896 exited with status 0xc0000005
--- Process 23444 thread 22692 exited with status 0xc0000005
--- Process 23444 thread 19000 exited with status 0xc0000005
--- Process 23444 thread 14112 exited with status 0xc0000005
--- Process 23444 thread 19868 exited with status 0xc0000005
--- Process 23444 thread 24252 exited with status 0xc0000005
--- Process 23444 thread 8376 exited with status 0xc0000005
--- Process 23444 thread 3560 exited with status 0xc0000005
--- Process 23444 thread 23356 exited with status 0xc0000005
--- Process 23444 thread 17896 exited with status 0xc0000005
--- Process 23444 thread 23512 exited with status 0xc0000005
--- Process 23444 thread 15652 exited with status 0xc0000005
--- Process 23444 exited with status 0xc0000005
--- Process 13396 exited with status 0xc0000005
`
</p>
</details>

Is there anything else I can provide to help resolve this?

omichel, 2022-06-03:
Instead of `make release`, can you `make debug` and launch Webots with `webots_debug.bat`? You should get a more useful stack.

Kreijstal, 2022-06-07:
Indeed this is way more useful, this is the output of webots_debug.bat:
```
Setup environment variables...
Generate the gdb input file...
Delete the gdb output file if exists
Starting Webots in debug mode...
warning: mincore\com\oleaut32\dispatch\ups.cpp(2122)\OLEAUT32.dll!00007FF909DC9DD6: (caller: 00007FF909DC91E9) ReturnHr(1) tid(5284) 8002801D Library not registered.
Delete the gdb input file...
```

Kreijstal, 2022-06-07:
[image]
This libary is found on system32 directory so I find it very strange or weird renaming it would solve the problem, I did it anyway to see if there was any success. And basically nothing changed. Is there a way I can give you a more thorough debug log?

omichel, 2022-06-07:
Can you type `ldd msys64/mingw64/bin/webots-bin.exe` to identify which one is used. On my system it is the one in `/c/Windows/System32/OLEAUT32.dll`.

Kreijstal, 2022-06-07:
Very interesting, ok I didn't tell you the truth, I renamed System/SysWOW64/OLEAUT32.dll because I assumed x64 application would resolve the dll there. after seeing this output:
```
 ldd webots-bin.exe
        ntdll.dll => /c/WINDOWS/SYSTEM32/ntdll.dll (0x7ff90bd10000)
        KERNEL32.DLL => /c/WINDOWS/System32/KERNEL32.DLL (0x7ff90b7e0000)
        KERNELBASE.dll => /c/WINDOWS/System32/KERNELBASE.dll (0x7ff9098e0000)
        ADVAPI32.dll => /c/WINDOWS/System32/ADVAPI32.dll (0x7ff909e90000)
        msvcrt.dll => /c/WINDOWS/System32/msvcrt.dll (0x7ff90b670000)
        sechost.dll => /c/WINDOWS/System32/sechost.dll (0x7ff90ba70000)
        RPCRT4.dll => /c/WINDOWS/System32/RPCRT4.dll (0x7ff90b460000)
        ole32.dll => /c/WINDOWS/System32/ole32.dll (0x7ff90bba0000)
        ucrtbase.dll => /c/WINDOWS/System32/ucrtbase.dll (0x7ff909bb0000)
        d3d9.dll => /c/WINDOWS/SYSTEM32/d3d9.dll (0x7ff8b3920000)
        combase.dll => /c/WINDOWS/System32/combase.dll (0x7ff90a3c0000)
        gdi32.dll => /c/WINDOWS/System32/gdi32.dll (0x7ff90bb70000)
        libgcc_s_seh-1.dll => /mingw64/bin/libgcc_s_seh-1.dll (0x7ff8e0bc0000)
        gdi32.dll => /c/Windows/System32/gdi32.dll (0x1778dee0000)
        USER32.dll => /c/WINDOWS/System32/USER32.dll (0x7ff90b2c0000)
        win32u.dll => /c/WINDOWS/System32/win32u.dll (0x7ff909d00000)
        GLU32.dll => /c/WINDOWS/SYSTEM32/GLU32.dll (0x7ff8df2b0000)
        libassimp-5.dll => /mingw64/bin/libassimp-5.dll (0x7ff861310000)
        win32u.dll => /c/Windows/System32/win32u.dll (0x1778dee0000)
        gdi32full.dll => /c/WINDOWS/System32/gdi32full.dll (0x7ff909420000)
        msvcp_win.dll => /c/WINDOWS/System32/msvcp_win.dll (0x7ff9097d0000)
        IPHLPAPI.DLL => /c/WINDOWS/SYSTEM32/IPHLPAPI.DLL (0x7ff908860000)
        OPENGL32.dll => /c/WINDOWS/SYSTEM32/OPENGL32.dll (0x7ff8df2e0000)
        libwinpthread-1.dll => /mingw64/bin/libwinpthread-1.dll (0x7ff8e0d10000)
        libminizip-1.dll => /mingw64/bin/libminizip-1.dll (0x7ff8de700000)
        libstdc++-6.dll => /mingw64/bin/libstdc++-6.dll (0x7ff864000000)
        zlib1.dll => /home/kreij/webots/msys64/mingw64/bin/zlib1.dll (0x7ff8b53b0000)
        libopenal-1.dll => /mingw64/bin/libopenal-1.dll (0x7ff863e50000)
        SHELL32.dll => /c/WINDOWS/System32/SHELL32.dll (0x7ff90aa00000)
        libpico.dll => /home/kreij/webots/msys64/mingw64/bin/libpico.dll (0x64340000)
        lua52.dll => /home/kreij/webots/msys64/mingw64/bin/lua52.dll (0x7ff888f60000)
        libfreetype-6.dll => /mingw64/bin/libfreetype-6.dll (0x7ff866070000)
        ode.dll => /home/kreij/webots/msys64/mingw64/bin/ode.dll (0x7ff85a320000)
        OIS.dll => /home/kreij/webots/msys64/mingw64/bin/OIS.dll (0x6e3c0000)
        OLEAUT32.dll => /c/WINDOWS/System32/OLEAUT32.dll (0x7ff909dc0000)
        openvr_api.dll => /home/kreij/webots/msys64/mingw64/bin/openvr_api.dll (0x7ff888f10000)
        Qt5Core.dll => /mingw64/bin/Qt5Core.dll (0x7ff859dd0000)
        Qt5Gui.dll => /mingw64/bin/Qt5Gui.dll (0x7ff859730000)
        Qt5Network.dll => /mingw64/bin/Qt5Network.dll (0x7ff863bc0000)
        Qt5OpenGL.dll => /mingw64/bin/Qt5OpenGL.dll (0x7ff888cf0000)
        WS2_32.dll => /c/WINDOWS/System32/WS2_32.dll (0x7ff90b740000)
        CRYPT32.dll => /c/WINDOWS/System32/CRYPT32.dll (0x7ff9095c0000)
        kernel.appcore.dll => /c/WINDOWS/SYSTEM32/kernel.appcore.dll (0x7ff907c90000)
        Qt5PrintSupport.dll => /mingw64/bin/Qt5PrintSupport.dll (0x7ff880580000)
        Qt5Qml.dll => /mingw64/bin/Qt5Qml.dll (0x7ff859320000)
        Qt5WebKit.dll => /mingw64/bin/Qt5WebKit.dll (0x7ff856c80000)
        comdlg32.dll => /c/WINDOWS/System32/comdlg32.dll (0x7ff90b590000)
        shcore.dll => /c/WINDOWS/System32/shcore.dll (0x7ff90a8a0000)
        SHLWAPI.dll => /c/WINDOWS/System32/SHLWAPI.dll (0x7ff90b1d0000)
        Qt5WebKitWidgets.dll => /mingw64/bin/Qt5WebKitWidgets.dll (0x7ff888680000)
        Qt5WebSockets.dll => /mingw64/bin/Qt5WebSockets.dll (0x7ff880620000)
        windows.storage.dll => /c/WINDOWS/SYSTEM32/windows.storage.dll (0x7ff9074e0000)
        dwmapi.dll => /c/WINDOWS/SYSTEM32/dwmapi.dll (0x7ff906b70000)
        Qt5Widgets.dll => /mingw64/bin/Qt5Widgets.dll (0x7ff8566b0000)
        Qt5Xml.dll => /mingw64/bin/Qt5Xml.dll (0x7ff8723d0000)
        WINMM.dll => /c/WINDOWS/SYSTEM32/WINMM.dll (0x7ff8ec390000)
        libbz2-1.dll => /mingw64/bin/libbz2-1.dll (0x7ff8e0ce0000)
        libharfbuzz-0.dll => /mingw64/bin/libharfbuzz-0.dll (0x7ff865760000)
        libpng16-16.dll => /mingw64/bin/libpng16-16.dll (0x7ff866600000)
        libbrotlidec.dll => /mingw64/bin/libbrotlidec.dll (0x7ff8de6e0000)
        XINPUT9_1_0.dll => /home/kreij/webots/msys64/mingw64/bin/XINPUT9_1_0.dll (0x7ff8b96d0000)
        DINPUT8.dll => /c/WINDOWS/SYSTEM32/DINPUT8.dll (0x7ff866240000)
        libdouble-conversion.dll => /mingw64/bin/libdouble-conversion.dll (0x7ff888090000)
        libicuin69.dll => /mingw64/bin/libicuin69.dll (0x7ff863620000)
        MPR.dll => /c/WINDOWS/SYSTEM32/MPR.dll (0x7ff8ea960000)
        libicuuc69.dll => /mingw64/bin/libicuuc69.dll (0x7ff863a00000)
        NETAPI32.dll => /c/WINDOWS/SYSTEM32/NETAPI32.dll (0x7ff8f2150000)
[... 30 lines cut ...]
        libicudt69.dll => /mingw64/bin/libicudt69.dll (0x7ff853210000)
        libwoff2common.dll => /mingw64/bin/libwoff2common.dll (0x6ec00000)
        libiconv-2.dll => /mingw64/bin/libiconv-2.dll (0x7ff860280000)
        liblzma-5.dll => /mingw64/bin/liblzma-5.dll (0x7ff861ec0000)
[... cut ...]
the new debug output is:
```
$ ./webots_debug.bat
Setup environment variables...
Generate the gdb input file...
Delete the gdb output file if exists
Starting Webots in debug mode...
warning: Could not load shared library symbols for C:\WINDOWS\System32\OLEAUT32.dll.
Do you need "set solib-search-path" or "set sysroot"?
warning: mincore\com\oleaut32\dispatch\ups.cpp(2122)\OLEAUT32.dll!00007FF909DC9DD6: (caller: 00007FF909DC91E9) ReturnHr(1) tid(61d0) 8002801D Library not registered.
Delete the gdb input file...
```

omichel, 2022-06-08:
OK, so it seems that the DLL used by Webots is the correct one.
I did the same as you and I am getting the exact same list of DLLs as you except that I do not get any duplicate for:
```
  gdi32.dll => /c/WINDOWS/System32/gdi32.dll
  win32u.dll => /c/WINDOWS/System32/win32u.dll
```
Which are duplicate in the list you provided. However, I am not sure if this has an impact on the result.

By searching the web, I found a possible cause for your problem at https://stackoverflow.com/a/45674648/810268
Is your PC up-to-date? I guess upgrading your `oleaut32.dll` should fix the problem. Mine is version 10.0.19041.985 (you can check it by opening the properties of the file:

[image]

What is your version?

Kreijstal, 2022-06-10:
[image]
Please excuse the delay, I have the same version as you do. I'll try with another machine, to see what is up.

Kreijstal, 2022-06-14:
solved:
Solution: WEBOTS_HOME was using mingw path and not a windows path.

closed 2022-06-14 by Kreijstal

omichel, 2022-06-14:
I don't understand why this was causing such an unrelated error...

omichel, 2022-06-14:
Maybe we could add some test somewhere in Webots to prevent this kid of nasty problem.