cyberbotics/webots #4618: Setting recognition colors does not apply until webots is restarted

Gabry993, 2022-06-09:
When setting a `recognitionColors` for a solid, the color does not appear in the camera frame (with recognition node, segmentation enabled, and both `recognitionEnable` and `enableRecognitionSegmentation` set from the script), while the bounding box immediately appears.
If I save the scene and then restart webots (not just the simulation), then the segmentation color is properly displayed as well.

I would like to automatically set the `recognitionColors` for a bunch of objects from a script when starting the simulation. I know the code to do this is working because I can see that the field is properly populated and, again, if I save the scene and restart webots, I can see the color displayed properly. 
However, the issue appears even if I set the field manually before starting the simulation: am I missing some initialization step that occurs when restarting, or should this work already?
Thank you!

stefaniapedrazzi, 2022-06-09:
Currently, the color used for the segmentation image is set only when an object is created.
This is why changing it after the world has been loaded doesn't work.
I'm just fixing the code to make it work and it would be available soon in the nightly build and in the next release.

A trick to workaround this missing functionality is to first change the field, then copy the node and delete the original.
It should be possible to achieve this exporting the node to string with [`Supervisor.exportString`](https://www.cyberbotics.com/doc/reference/supervisor?tab-language=python#wb_supervisor_node_export_string) and then adding a new copy with [`Supervisor.importMFNodeFromString`](https://www.cyberbotics.com/doc/reference/supervisor?tab-language=python#wb_supervisor_field_import_mf_node_from_string).

Gabry993, 2022-06-09:
Thank you for your prompt response.
I've implemented the workaround and it works for one run, then when I restart the simulation, all re-imported objects are gone.
I tried to save the world after re-importing the objects, but calling `Supervisor.saveWorld()` gives me this error:
```Error: wb_supervisor_world_save() called with a NULL 'filename' argument.```
From the [doc](https://cyberbotics.com/doc/reference/supervisor?tab-language=python#wb_supervisor_world_save) I understood this would work (also, I tried to set explicitly both `file` and `filename` argument, but I get `TypeError: worldSave() got an unexpected keyword argument 'file'`, so probably there is an error in the doc? Should I open a new issue for this?

Here is a code snippet of the implemented workaround:
```python
robot = Supervisor()
objects_node = robot.getFromDef("Objects")
# iterate over the children of Objects group to set recognition colors
for i in range(objects_node.getField('children').getCount()):
    solid = objects_node.getField('children').getMFNode(i)
    solid.getField('recognitionColors').insertMFColor(0, [0., 1., 0.])
    solid_string = solid.exportString()
    objects_node.getField('children').removeMF(i)
    objects_node.getField('children').importMFNodeFromString(i, solid_string)
robot.worldSave() # also robot.worldSave(file=None) and robot.worldSave(filename=None) fail
```

stefaniapedrazzi, 2022-06-09:
The documentation is correct and should work.
I tested and it works correctly for me with Python 3.10.
Note that `Supervisor.saveWorld()` is deprecated and make sure that you use `Supervisor.worldSave()`.

Alternatively you could overwrite the file by explicitly passing the current world path using the instruction:
```python
robot.worldSave(robot.getWorldPath())
```

Gabry993, 2022-06-09:
I'm using python 3.8.10, Webots R2022a (December 2021, maybe it's not the latest?) and calling `Supervisor.worldSave()` (sorry, I've mixed the 2 in the previous comment).
I don't know why `robot.worldSave()` gives me that error: looking at the code here on github, the error print is not even there. I've installed Webots on Ubuntu 20 from apt, maybe I've missed some updates?
Passing the file path as suggested works and the workaround functions correctly, thank you!

stefaniapedrazzi, 2022-06-13:
This issue with `robot.worldSave` in Webots <= R2022a has been fixed in #4140.
The fix will be available in the next release of Webots R2022b coming soon or you can download a nightly build that already contains it from this page:
https://github.com/cyberbotics/webots/releases/tag/nightly_10_6_2022

PR #4619 (2022-06-13): Fix segmentation color change
Fix #4618: when `Solid.recognitionColors` changes, the descendant meshes (WbGeometry, WbCadShape and WbSkin) needs to be notified to update their segmentation color.
closed 2022-06-13 by stefaniapedrazzi