cyberbotics/webots #4650: Redundant copy of libode.so

ygoumaz, 2022-06-14:
When building Webots develop branch from sources after a `make cleanse`, the ODE library `libode.so` is compiled and copied to `WEBOTS_HOME/lib/webots`. This should only be the case if `libode.so` is not already located in `WEBOTS_HOME/lib/webots`. If multiple `make`s are done consecutively, libode.so is always copied.
```
# copying libode.so to /home/yannick/Documents/cyberbotics/webotsdev/lib/webots/libode.so
```
**Steps to Reproduce**
1. Install Webots development envrionment (develop branch) or `make cleanse` in an already existing one.
2. `make` -> ode library is compiled, linked and copied.
3. `make` again -> ode library is copied once again but it should not.

I believe the problem was introduced in #4543 but I am not 100% sure.

BenjaminDeleze, 2022-06-15:
It should also be the case if `libode.so` has changed

stefaniapedrazzi, 2022-06-16:
Just for information, the reason why the copy was not done immediately to the `lib`/webots folder is that this way it is possible to have at the same time the `release` and `debug` version of the ode library and switch between them without having to recompile (if anything changed).

The same mechanism is also used for webots.

omichel, 2022-06-16:
> The same mechanism is also used for webots.

I believe this is not exact. For webots, we keep the debug object files in the `src/webots/build/debug` folder and the release object files in the `src/webots/build/release` folder. A similar mechanism applies to ode. However, for ode we have also two versions of the `libode.dll` inside these build folders whereas with webots, we don't link webots in the build folder and the webots binary is linked directly in the directory from where it is executed.

I believe it would be better to build the ode.dll file directly in the folder where it is used and thus avoid a copy instead of building two versions of ode.dll.

When we target debug instead of release, a simple link will be performed against the debug object files and a new DLL binary will be generated in the folder where it is used, so switching from debug to release and back is fast.

I believe copies in a build process should be avoided as much as possible and the proposed mechanism seems better than the current one which performs useless copies on every global make.

PR #4661 (2022-06-17): Fix redundant copy of libode.so
Fix #4650 

This PR is changing the target directory when linking the libode.so shared library. This way no copy is needed anymore.
closed 2022-06-17 by omichel