cyberbotics/webots #4665: Conflict between cyberbotics.com and webots-js CSS

ThomasOliverKimble, 2022-06-16:
Conflict between cyberbotics.com `CSS` and progress bar `CSS` causing bug in appearance (can be seen [here](https://cyberbotics.com/doc/guide/aibo-ers7?version=develop) for example):

[image]

BenjaminDeleze, 2022-06-16:
The main issue seems to come from the `text-align: justify;` property

ThomasOliverKimble, 2022-06-16:
> The main issue seems to come from the `text-align: justify;` property

Indeed, however I found an easier solution by switching the `<p>` elements to `<div>`, I am finishing up this in #4666.

BenjaminDeleze, 2022-06-16:
Fine, but what if on a user's website there is rule with `div { text-align: justify/left/right/...;}`, will it work?

ThomasOliverKimble, 2022-06-16:
> Fine, but what if on a user's website there is rule with `div { text-align: justify/left/right/...;}`, will it work?

I just tested this by implementing this rule in the streaming viewer (in `style.css`) and it does not affect the progress bars properties. If they are a `div`, I believe the class name or id is enough to set new properties for the element. Let me know if you find anything different.

PR #4666 (2022-06-17): Fix doc progress bar
**Description**
Fixes #4665 and makes progress bar responsive for smaller/mobile screens.

[image]
closed 2022-06-17 by BenjaminDeleze