cyberbotics/webots #4685: new project webot wizard wont run properly

oktaspc, 2022-06-18:
**Describe the Bug**
i just download webots R2022a when i open it and wanna make new project. wizard> new project directory
this is happened :

**screenshoot**
[image]

**System**
Linux Ubuntu 22.04

note : 
i installed it 2 times and still happaned
first time try installed it trough terminal
second time download it using snap
internet connection stable. i can open the source using my browser.

**Expected behavior**
WARNING: RectangleArena (PROTO) > Parquetry (PROTO) > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/parquetry/chequered_parquetry_base_color.jpg: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/parquetry/chequered_parquetry_base_color.jpg
WARNING: RectangleArena (PROTO) > Parquetry (PROTO) > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/parquetry/chequered_parquetry_roughness.jpg: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/parquetry/chequered_parquetry_roughness.jpg
WARNING: RectangleArena (PROTO) > Parquetry (PROTO) > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/parquetry/chequered_parquetry_normal.jpg: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/parquetry/chequered_parquetry_normal.jpg
WARNING: RectangleArena (PROTO) > Parquetry (PROTO) > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/parquetry/chequered_parquetry_occlusion.jpg: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/parquetry/chequered_parquetry_occlusion.jpg
WARNING: RectangleArena (PROTO) > BrushedAluminium (PROTO) > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/brushed_aluminium/brushed_aluminium_base_color.jpg: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/brushed_aluminium/brushed_aluminium_base_color.jpg
WARNING: RectangleArena (PROTO) > BrushedAluminium (PROTO) > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/brushed_aluminium/brushed_aluminium_roughness.jpg: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/brushed_aluminium/brushed_aluminium_roughness.jpg
WARNING: RectangleArena (PROTO) > BrushedAluminium (PROTO) > ImageTexture: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/brushed_aluminium/brushed_aluminium_normal.jpg: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/appearances/protos/textures/brushed_aluminium/brushed_aluminium_normal.jpg
WARNING: TexturedBackground (PROTO): Cannot retrieve 'https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/default/worlds/textures/cubic/mountains_back.png': Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/default/worlds/textures/cubic/mountains_back.png: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/default/worlds/textures/cubic/mountains_back.png
WARNING: TexturedBackground (PROTO): Cannot download leftIrradianceUrl: Cannot download https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/default/worlds/textures/cubic/mountains_back.hdr: Error opening https://raw.githubusercontent.com/cyberbotics/webots/R2022a/projects/default/worlds/textures/cubic/mountains_back.hdr

omichel, 2022-06-18:
Could you please with the latest [nightly build](https://github.com/cyberbotics/webots/releases) of Webots R2022b? It should work with this version.

ygoumaz, 2022-06-20:
Ubuntu 22.04 is not supported by the latest stable release of Webots (R2022a). As suggested by @omichel, you should use the latest nightly build of Webots (R2022b).

ahujan456, 2023-03-11:
I am also getting the same error on windows os anyone please help

omichel, 2023-03-13:
> I am also getting the same error on windows os anyone please help

Which version of Webots are you using? Are you behind some firewall?