cyberbotics/webots #4702: Inertia matrix point of calculation

ShuffleWire, 2022-06-21:
**Describe the Bug**
It seems that, contrary to what say the documentation, the matrix of inertia (MoI) we provide in the Physic node is computed on the origin of the object, not on the center of mass (CoM).
I'm kinda scared to say something completely off, because the MoI computation seems to give good results in all Webots projects, so I don't know, I could have missed some information...

I got some observation that got me to that point :  
Considering this world : 

<details>
<summary>World</summary>
<pre>
#VRML_SIM R2022b utf8
WorldInfo {
  lineScale 10
}
Viewpoint {
  orientation 0.18483272608525197 0.34484097093526966 -0.9202834172863895 1.2065546190350083
  position -8.08416062735125 16.7740818030343 8.768359541015466
}
TexturedBackground {
}
TexturedBackgroundLight {
}
Robot {
  children [
    DEF t Transform {
      children [
        Shape {
          appearance PBRAppearance {
          }
          geometry Sphere {
          }
        }
      ]
    }
  ]
  boundingObject USE t
  physics Physics {
    density -1
    mass 1
  }
}
</pre>
</details>

**Steps to Reproduce**
1. Open the world, note the Robot CoM is (0,0,0), MoI is [0.4,0.4,0.4, 0,0,0] 
That is correct
2. Change the x coordinate of the transform "t" from 0 to 1
3. Observe that the CoM of the robot change to (1,0,0)
That is correct.
But observe that the MoI have changed as well to [0.4,1.4,1.4, 0,0,0], and those values correspond to the MoI computed "as if" the CoM was at the origin.
If the inertia matrix were effectively computed at the CoM, it should not change.
4. Additionally, set whatever value for the CoM (for instance (2,0,0)), and observe that the MoI stay at [0.4,1.4,1.4, 0,0,0]
5. I've check to code in ode that is called to check to mass validity https://github.com/cyberbotics/webots/blob/4afa98838cde2bdb2eba4c54cc8593298c45dcf8/src/ode/ode/src/mass.cpp#L43
Just to get a feeling of how the MoI values are actually interpreted.
The comment make me feel that to value of inertia matrix are at a "reference point" as they say, which could be the origin. Anyway, if the matrix was given at the CoM, I don't think we need the operation they are doing (I + mass*crossmat(c)^2) 
But really, I'm not sure at all to interpret that correctly (given that my English is kinda broken, sometimes xD)

That let me think that the MoI is actually computed / displayed at the relative origin of the object, and not at his CoM

**Expected behavior**
Documentation (https://cyberbotics.com/doc/reference/physics?version=develop#field-summary) states that "The inertia matrix is defined with respect to the centerOfMass of the Solid."
I was thinking that maybe the actual display means something different from the input describes here.
If not, I think there are some fixes to be applied, maybe simply change the documentation ?

**System**
 - Operating System: Debian 11
 - Webots: 2022b

Hope I could help in finding a bug :)

ShuffleWire, 2022-07-21:
Also, please note that the MoI is always displayed in the local coordinate system of the object, even if we select "absolute" in the "Mass" tab. However, it could be that it actually apply only to the CoM, given that this toggle is just next to the CoM label.

But I do think that it could be nice to get the MoI in the global origin as well, just to stay consistent with the CoM.

Also, we display the CoM in the 3D view using a 3D cross which as simply axis along XYZ axis. But I do think that using the principals axis of the matrix to draw the 3D cross could bring more info without compromising the visual representation on the CoM location.

RuchaoWang, 2023-12-25:
I used the "Group" tools of webots to build model and did not encounter such problems. However, using Solidworks to build model and export the .stl format has the same problem as you mentioned, and through comparison, it is found that the standards for the product of inertia they use are different. I assume that SolidWorks is accurate and can be calculated through formulas.