cyberbotics/webots #4711: Improve Bitcraze Crazyflie

omichel, 2022-06-22:
Here are some ideas to improve the Bitcraze Crazyflie simulation:

- [ ] Implement the fastHelix / slowHelix as in the [Mavic2Pro model](https://github.com/cyberbotics/webots/blob/master/projects/robots/dji/mavic/protos/Mavic2Pro.proto) for a [better rendering the helices](https://www.youtube.com/watch?v=-hJssj_Vcw8).
- [ ] Add altitude control to the keyboard controller.
- [ ] Improve the world file with obstacles and a more real world appearance, it could for example implement a simple [apartment](https://cyberbotics.com/doc/guide/samples-environments#apartment-wbt).

knmcguire, 2022-06-22:
Thanks! This is a nice list of to-dos. I've added some additional items that we are planning to do as well.

Klark007, 2023-04-14:
Looking at the sensor list it seems that the accelerometer, which is part of the crazyflie's real sensors, is not in the Proto. This could also be added to the above list

knmcguire, 2023-04-17:
Good point! Currently we haven't included that since it is mostly used for state estimate filter and that is currently replaced by the 'gps', 

So many things to fix... I'm hoping that I'll be able to do this before the 2023b release but not sure when that is planned

omichel, 2023-04-17:
We plan to release Webots R2023b on June 1st 2023.

ShuffleWire, 2023-09-07:
Could we close this issue, the PR being merged ?

omichel, 2023-09-07:
Sure.

closed 2023-09-07 by omichel

knmcguire, 2023-09-08:
Hi! technically there were some more improvements in the pipeline, but the current state of the crazyflie is pretty stable at the moment. so I'm okay with closing this. At least all the improvements that @omichel had proposed initially are taken care off 😄