cyberbotics/webots #4716: Copying a robot blocks the simulation

stefaniapedrazzi, 2022-06-22:
Steps to reproduce:
1) open `accelerometer.wbt`
2) run simulation
3) copy `MyBot` robot
-> simulation blocks for long time (many minutes)

After some time the following errors are printed in the console:
```
Cannot connect to Webots instance on socket "/tmp/webots-1234/ipc/MyBot%281%29/intern", retrying in 0 second...
Cannot connect to Webots instance on socket "/tmp/webots-1234/ipc/MyBot%281%29/intern", retrying in 1 second...
Cannot connect to Webots instance on socket "/tmp/webots-1234/ipc/MyBot%281%29/intern", retrying in 2 seconds...
Cannot connect to Webots instance on socket "/tmp/webots-1234/ipc/MyBot%281%29/intern", retrying in 3 seconds...
Cannot connect to Webots instance on socket "/tmp/webots-1234/ipc/MyBot%281%29/intern", retrying in 4 seconds...
Cannot connect to Webots instance on socket "/tmp/webots-1234/ipc/MyBot%281%29/intern", retrying in 5 seconds...
Cannot connect to Webots instance on socket "/tmp/webots-1234/ipc/MyBot%281%29/intern", retrying in 6 seconds...
Cannot connect to Webots instance on socket "/tmp/webots-1234/ipc/MyBot%281%29/intern", retrying in 7 seconds...
Cannot connect to Webots instance on socket "/tmp/webots-1234/ipc/MyBot%281%29/intern", retrying in 8 seconds...
Cannot connect to Webots instance on socket "/tmp/webots-1234/ipc/MyBot%281%29/intern", retrying in 9 seconds...
Cannot connect to Webots instance on socket "/tmp/webots-1234/ipc/MyBot%281%29/intern", retrying in 10 seconds...
Cannot connect to Webots instance on socket "/tmp/webots-1234/ipc/MyBot%281%29/intern", retrying in 11 seconds...
Giving up...
WARNING: 'accelerometer' controller exited with status: 1.
```

**Expected behavior**
Webots should not block and the old and new robots should move immediately.

omichel, 2022-06-22:
I cannot reproduce this problem on Windows. I assume you observed it on Linux, right?

stefaniapedrazzi, 2022-06-22:
Yes, on Ubuntu 22.04 both on a clean system with the snap and on my development system.

omichel, 2022-06-22:
I cannot reproduce it on Ubuntu 20.04 under my development environment.
I will try Ubuntu 22.04 now...

omichel, 2022-06-27:
Fixed in #4748.

PR #4748 (2022-06-27): Fixed hang after copy/paste of robots on Linux
Fixes #4716.