cyberbotics/webots #4717: Broken viewpoint rotation around selected object

stefaniapedrazzi, 2022-06-22:
If an object is selected when rotating the viewpoint, the rotation should be centered at selected object position.
However the rotation center in R2022b still seems to be the viewpoint center.

Not sure if this is an issue of the VM (Ubuntu 22.04 VirtualBox) only.
Tested with 21.06 nightly build snap package.
I will further investigate

ShuffleWire, 2022-06-22:
(I hope we are talking about the same thing). I could reproduce the issue using the develop branch today, on a Debian 11. 
However, I have just checked and I got the same behavior with R2021b, is that the same on your end ?

omichel, 2022-06-23:
Rotating the viewpoint is not based on the selection, but on the picked object. Regardless of the selection, the rotation should take place around the picked point (intersection between the ray from the viewpoint going forward and the first object encountered). I tested it on Ubuntu 22.04 (development environment) under VMware and it works as expected. The bug you experienced may be caused by an OpenGL bug of VirtualBox related to the picking.

stefaniapedrazzi, 2022-06-23:
Yes, I also think it is related to some weird behavior on VirtualBox because I cannot reproduce it on my native system even if I install Webots using the snap package.

However, it is strange because picking seems to work works correctly:
* on click the correct object is always selected
* rotating the viewpoint by clicking on the background always works correctly

omichel, 2022-06-23:
I believe this is not a regression.