cyberbotics/webots #4725: Textures not found in custom world created with snap package on Ubuntu 22.04

stefaniapedrazzi, 2022-06-22:
Steps to reproduce:
1) open `acceleremeter.wbt`
2) create a new project with the wizard
3) stop simulation
4) add an E-puck robot
5) move viewpoint to the robot
6) save world
7) close webots
8) restart webots
-> errors about textures of TexturedBackground not found

It seems that there are some conflicts, because after this any world containing the TextureBaground will throw errors.
However, if you open a default world (for example `accelerometer.wbt`) and restart Webots no error is displayed.

ad-daniel, 2022-06-22:
Can you post the resulting custom world? Specifically, what's the resulting EXTERNPROTO from the save

stefaniapedrazzi, 2022-06-22:
```
#VRML_SIM R2022b utf8

EXTERNPROTO "https://raw.githubusercontent.com/cyberbotics/webots/96decb2fc65a6d0917e1d3f34365d6e17c1600e3/projects/robots/gctronic/e-puck/protos/E-puck.proto"

WorldInfo {
}
Viewpoint {
  orientation -0.5773502691896257 0.5773502691896257 0.5773502691896257 2.0944
  position -0.0003612746519147755 0.0009117271480573368 0.2911036105363358
}
TexturedBackground {
}
TexturedBackgroundLight {
}
E-puck {
}
```

stefaniapedrazzi, 2022-06-22:
I see.
The problem is the that the wizard adds the default PROTOs without setting them in the world.

PR #4731 (2022-06-27): Fix declaration of EXTERNPROTO in new world
Fix #4725:

generate new world instead of copying the `resources/projects/worlds/empty.wbt` file when creating a new project from the wizard and add the required EXTERNPROTO declarations.
closed 2022-06-27 by stefaniapedrazzi