cyberbotics/webots #4726: Forest PROTO not found aftre relocating highway_driving.wbt

stefaniapedrazzi, 2022-06-22:
Steps to reproduce with snap installation on Ubuntu 22.04:
1) open `highway_driving.wbt`
2) move viewpoint, save and relocate project
3) reload the simulation
-> errors
```
ERROR: JavaScript error: could not open file: /home/stefi/Desktop/highway_driving2/worlds/forest/highway_overtake/1.forest.
WARNING: DEF WEBOTS_VEHICLE0 LincolnMKZ (PROTO): The 'automobile' robot window library has not been found.
```

(the second warning should be related to #4719)

stefaniapedrazzi, 2022-06-22:
This is simply because when relocating the forest folder in worlds is not copied.

ad-daniel, 2022-06-28:
This issue only mentions 2 warnings, do you recall if you had any other when relocating the project?
I've fixed the forest relocation, but to me there's a another warning concerning the fact it cannot find the controller also. Might be a regression we introduced these past days

stefaniapedrazzi, 2022-06-28:
I just had this two warnings, the controllers were found and worked correctly apart from the `highway_driving_benchmark` that doesn't run from Webots because in the snap the Python `shapely` module is not found.
(I didn't test with extern controller).

omichel, 2022-06-29:
I believe both issues regarding the forest relocation and automobile robot window not found were resolved in #4768 and #4790 respectively.

closed 2022-06-29 by omichel