cyberbotics/webots #4784: Can't convert a Robot with connector to Solid

ShuffleWire, 2022-06-28:
**Describe the Bug**
A Robot with a Connector in children can't be converted to Solid through Right Click / Transform To...

**Steps to Reproduce**
1. Open empty world
2. Add a Robot
3. Observe that  "Right Click / Transform To..." show the Solid option
4. Add a Connector in children
5. Observe that  "Right Click / Transform To..." doesn't show the Solid option
6. Save the file, open it in text editor, change "Robot" to "Solid"
7. Open in Webot again, everything is fine, Solid is used

That fine, because a Solid could have a "passive" connector.

**Expected behavior**
We should be able to convert "Robot" to "Solid" through GUI

**System**
 - Operating System: Debian 11
 -  Webot R2022a

ShuffleWire, 2022-07-12:
Short answer : should WbConnector class be changed from WbSolidDevice to WbDevice ?

Long answer :

I might get to the source of the issue : reading the content of the function ;:
bool WbNodeUtilities::hasSolidChildren(const WbNode *node)

I've figured out that we do a cast to WbSolid. However, the WbConnector class derive from WbSolidDevice (and so from WbSolid), hence trigger the return true, and block the conversion.

I've tried with IMU, LED, GPS node, and same behavior, and they also derive from the WbSolidDevice class.

Is there anything we can do with that, because converting a Robot with a Connector in it is the only case where the conversion make sense (contrary to a Robot with IMU, LED, GPS .., which can't because a basic Solid with that those node in it) 

Should the type of the Connector class be changed or will it impact others stuff ?

ShuffleWire, 2022-07-12:
other question : always in bool WbNodeUtilities::hasSolidChildren(const WbNode *node) , should 
```if (it.hasNext())```
be remplaced with 
```while (it.hasNext())```
?
Currently, if I add a Shape (a node which doesn't derive from Solid) before any other node (whenever they are Solid-derivative and Robot-to-Solid convertible), the option "Transform to ... Solid" on the Robot show up, even if it should not.

stefaniapedrazzi, 2022-07-14:
> Short answer : should WbConnector parent class be changed from WbSolidDevice to WbDevice ?

No. It is correct that `WbConnector` inherits from `WbSolidDevice`.

closed 2022-07-21 by stefaniapedrazzi