cyberbotics/webots #4811: Replace "webots://" URLs with "/~/"

omichel, 2022-06-30:
In #4803, we introduced a special URL scheme for accessing resources belonging to the local Webots installation.
This URL was defined with a `/~/` prefix to indicate this is a local Webots resource.
For example, we refer to the automobile robot window with the following URL: `/~/projects/vehicles/plugins/robot_windows/automobile` where as we can refer to a local file with the following URL: `plugins/robot_windows/my_robot_window`.

I am wondering whether we should apply the same formalism and replace `webots://` URLs by `/~/` also in world and PROTO files.
That would be more consistent.

What do you think?

BenjaminDeleze, 2022-07-04:
For me it is clearer that `webots://` correspond to `$WEBOTS_HOME` than `/~/` that seems more like the `$HOME`

omichel, 2022-07-04:
> For me it is clearer that `webots://` correspond to `$WEBOTS_HOME` than `/~/` that seems more like the `$HOME`

I agree, but the problem is that `webots://` won't work for robot window URLs and I believe it is fundamentally wrong to specify some path-related information in the protocol part of a URL. An alternative to `/~/` could be `/~webots/` to make it clearer that we refer to `$WEBOTS_HOME`, or we maybe `/$WEBOTS_HOME/` which would be even more straightforward, but looks a bit odd for a URL?

omichel, 2022-07-04:
This should be fixed in #4822.

closed 2022-07-04 by omichel