cyberbotics/webots #4823: Text of EPHERMAL PROTO not really readable

BenjaminDeleze, 2022-07-04:
As you can see on the picture, the description text before the `Insert new` button is cropped and the no slider is visible (even so we can scroll the text).

I have the same problem in both night and classic theme
[image]

BenjaminDeleze, 2022-07-04:
The problem has been introduced in #4701 

Before:
[image]

After:
[image]

I can either try to restore the original look specifically for this text only or put the `Insert new` button lower to let the room to display the fulltext.

What do you think?

ad-daniel, 2022-07-04:
I think it might also depend on how this https://github.com/cyberbotics/webots/pull/4826 is addressed, based on the available space and what not

PR #4828 (2022-07-04): Fixed font in ephemeral proto warning
Fixes #4823:

[image]

(Tested on Windows and Linux with various themes)
closed 2022-07-04 by omichel