cyberbotics/webots #4848: Can I download prototypes and maps offline?

sxbug, 2022-07-06:
After installing the latest webots（night build） and opening the example program, it is all black and can not see anything, Console indicating that many prototypes and maps have not been downloaded, but I could open prototypes with Edge  (Need try  several times). Is there a way to download them offline?

thank you

ad-daniel, 2022-07-06:
What do you mean by prototypes and maps? But yes, with all nightly builds there's an `assets-R2022b.zip` included which you can put in your cache following these instructions: https://cyberbotics.com/doc/guide/installation-procedure?version=develop&tab-os=windows#asset-cache-download

sxbug, 2022-07-07:
Thanks。I See

closed 2022-07-07 by ad-daniel