cyberbotics/webots #4875: Spawning of multiple objects from a PROTO results in a crash

ad-daniel, 2022-07-12:
**Describe the Bug**

Spawning of multiple objects from a PROTO results in a crash.

```
#VRML_SIM R2022a utf8
# template language: javascript

PROTO MultiSpawner [
  field SFInt32 numberOfRobots 5
  field SFVec2f floorSize      10 10
]
{
  %<
    import * as wbrandom from 'wbrandom.js';

    const size = fields.floorSize.value;
    const robots = fields.numberOfRobots.value;
  >%
  Group {
    children [
    %< for (let j = 0; j < robots; ++j) { >%
      E-puck {
        translation %<= wbrandom.integer(0, size.x) >% %<= wbrandom.integer(0, size.y) >% 0.5
      }
    %< } >%
    ]
  }
}
```

The issue doesn't appear to be related to JavaScript, the following PROTO also crashes:
```
#VRML_SIM R2022a utf8

PROTO MultiSpawner [
]
{
  Group {
    children [
      E-puck {
      }
      E-puck {
      }
      E-puck {
      }
      E-puck {
      }
      E-puck {
      }
      E-puck {
      }
      E-puck {
      }
      E-puck {
      }
      E-puck {
      }
      E-puck {
      }
      E-puck {
      }
      E-puck {
      }
      E-puck {
      }
      E-puck {
      }
      E-puck {
      }
      E-puck {
      }
      E-puck {
      }
      E-puck {
      }
      E-puck {
      }
      E-puck {
      }
      E-puck {
      }
      E-puck {
      }
      E-puck {
      }
      E-puck {
      }
    ]
  }
}
```

The type of node doesn't seem to have an effect either, swapping the E-puck for an Apple crashes as well.

stefaniapedrazzi, 2022-07-12:
I can reproduce a crash due to the Speaker device when using the `E-puck` PROTO.
But it doesn't crash for me when using the `Apple` PROTO.
I will fix the first crash and try again with other PROTO objects.

stefaniapedrazzi, 2022-07-12:
The crashes are fixed in #4878 for me.
However #4878 just fixes issues with robots, so if there were issues also when using only `Apple` PROTO instead of the `E-puck` PROTO these issues should still be there.
But I cannot reproduce them.

closed 2022-07-12 by stefaniapedrazzi

ad-daniel, 2022-07-12:
I can confirm I can't reproduce the issue on master with non-robot objects either. I mostly did those tests on R2022a so perhaps it was exclusive to that version.