cyberbotics/webots #4881: Compiling development branch linking issues

Kreijstal, 2022-07-12:
**Describe the Bug**
A clear and concise description of what the bug is.

**Steps to Reproduce**
1. Go to development branch
2. Install dependencies and dev dependencies
3. make
4. See error

**Additional context**
Add any other context about the problem here.

This is the error message from make
```
# compiling build/release/WbZoom.moc.cpp
# compiling build/release/WbVirtualRealityHeadset.moc.cpp
# compiling gui/webots.o
# linking  ../../msys64/mingw64/bin/webots-bin.exe
C:/Users/kreij/scoop/apps/msys2/2022-01-28/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: build/release/WbAboutBox.o:WbAboutBox.cpp:(.text+0xc9): undefined reference to `__imp__ZN7QString16fromAscii_helperEPKci'
C:/Users/kreij/scoop/apps/msys2/2022-01-28/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: build/release/WbAboutBox.o:WbAboutBox.cpp:(.text+0x156): undefined reference to `__imp__ZN11QGridLayoutC1Ev'
C:/Users/kreij/scoop/apps/msys2/2022-01-28/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: build/release/WbAboutBox.o:WbAboutBox.cpp:(.text+0x3a9): undefined reference to `__imp__ZNK5QDate8toStringERK7QString'
C:/Users/kreij/scoop/apps/msys2/2022-01-28/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: build/release/WbAboutBox.o:WbAboutBox.cpp:(.text+0x96d): undefined reference to `__imp__ZN10QArrayData10deallocateEPS_yy'
C:/Users/kreij/scoop/apps/msys2/2022-01-28/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: build/release/WbAboutBox.o:WbAboutBox.cpp:(.text+0x98d): undefined reference to `__imp__ZN10QArrayData10deallocateEPS_yy'
C:/Users/kreij/scoop/apps/msys2/2022-01-28/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: build/release/WbAboutBox.o:WbAboutBox.cpp:(.text+0x9ad): undefined reference to `__imp__ZN10QArrayData10deallocateEPS_yy'
C:/Users/kreij/scoop/apps/msys2/2022-01-28/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: build/release/WbAboutBox.o:WbAboutBox.cpp:(.text+0x9d0): undefined reference to `__imp__ZN10QArrayData10deallocateEPS_yy'
C:/Users/kreij/scoop/apps/msys2/2022-01-28/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: build/release/WbAboutBox.o:WbAboutBox.cpp:(.text+0x9f3): undefined reference to `__imp__ZN10QArrayData10deallocateEPS_yy'
C:/Users/kreij/scoop/apps/msys2/2022-01-28/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: build/release/WbAboutBox.o:WbAboutBox.cpp:(.text+0xa16): more undefined references to `__imp__ZN10QArrayData10deallocateEPS_yy' follow
C:/Users/kreij/scoop/apps/msys2/2022-01-28/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: build/release/WbAbstractAppearance.o:WbAbstractAppearance.cpp:(.text+0x22f): undefined reference to `__imp__ZN7QString16fromAscii_helperEPKci'
C:/Users/kreij/scoop/apps/msys2/2022-01-28/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: build/release/WbAbstractAppearance.o:WbAbstractAppearance.cpp:(.text+0x2ed): undefined reference to `__imp__ZN10QArrayData10deallocateEPS_yy'
C:/Users/kreij/scoop/apps/msys2/2022-01-28/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: build/release/WbAbstractAppearance.o:WbAbstractAppearance.cpp:(.text+0x305): undefined reference to `__imp__ZN10QArrayData10deallocateEPS_yy'
C:/Users/kreij/scoop/apps/msys2/2022-01-28/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: build/release/WbAbstractAppearance.o:WbAbstractAppearance.cpp:(.text+0x342): undefined reference to `__imp__ZN10QArrayData11shared_nullE'
C:/Users/kreij/scoop/apps/msys2/2022-01-28/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: build/release/WbAbstractAppearance.o:WbAbstractAppearance.cpp:(.text+0x3a2): undefined reference to `__imp__ZN10QArrayData11shared_nullE'
C:/Users/kreij/scoop/apps/msys2/2022-01-28/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: build/release/WbAbstractAppearance.o:WbAbstractAppearance.cpp:(.text+0x402): undefined reference to `__imp__ZN10QArrayData11shared_nullE'
C:/Users/kreij/scoop/apps/msys2/2022-01-28/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: build/release/WbAbstractAppearance.o:WbAbstractAppearance.cpp:(.text+0x486): undefined reference to `__imp__ZN10QArrayData11shared_nullE'
C:/Users/kreij/scoop/apps/msys2/2022-01-28/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: build/release/WbAbstractAppearance.o:WbAbstractAppearance.cpp:(.text+0x4f3): undefined reference to `__imp__ZN7QString16fromAscii_helperEPKci'
C:/Users/kreij/scoop/apps/msys2/2022-01-28/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: build/release/WbAbstractAppearance.o:WbAbstractAppearance.cpp:(.text+0x595): undefined reference to `__imp__ZN10QArrayData10deallocateEPS_yy'
C:/Users/kreij/scoop/apps/msys2/2022-01-28/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: build/release/WbAbstractAppearance.o:WbAbstractAppearance.cpp:(.text+0x6bc): undefined reference to `__imp__ZN10QArrayData10deallocateEPS_yy'
C:/Users/kreij/scoop/apps/msys2/2022-01-28/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: build/release/WbAbstractAppearance.o:WbAbstractAppearance.cpp:(.text+0x6e1): undefined reference to `__imp__ZN10QArrayData10deallocateEPS_yy'
C:/Users/kreij/scoop/apps/msys2/2022-01-28/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: build/release/WbAbstractAppearance.o:WbAbstractAppearance.cpp:(.text+0x7dd): undefined reference to `__imp__ZN10QArrayData10deallocateEPS_yy'
C:/Users/kreij/scoop/apps/msys2/2022-01-28/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: build/release/WbAbstractAppearance.o:WbAbstractAppearance.cpp:(.text+0x7f9): undefined reference to `__imp__ZN10QArrayData10deallocateEPS_yy'
C:/Users/kreij/scoop/apps/msys2/2022-01-28/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: build/release/WbAbstractAppearance.o:WbAbstractAppearance.cpp:(.text$_ZN10WbSFStringD0Ev[_ZN10WbSFStringD0Ev]+0x59): more undefined references to `__imp__ZN10QArrayData10deallocateEPS_yy' follow
C:/Users/kreij/scoop/apps/msys2/2022-01-28/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: build/release/WbAbstractCamera.o:WbAbstractCamera.cpp:(.text+0x624): undefined reference to `__imp__ZN9QListData6detachEi'
C:/Users/kreij/scoop/apps/msys2/2022-01-28/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: build/release/WbAbstractCamera.o:WbAbstractCamera.cpp:(.text+0x67d): undefined reference to `__imp__ZN9QListData7disposeEPNS_4DataE'
C:/Users/kreij/scoop/apps/msys2/2022-01-28/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: build/release/WbAbstractCamera.o:WbAbstractCamera.cpp:(.text+0x758): undefined reference to `__imp__ZN9QListData6detachEi'
C:/Users/kreij/scoop/apps/msys2/2022-01-28/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: build/release/WbAbstractCamera.o:WbAbstractCamera.cpp:(.text+0x7b5): undefined reference to `__imp__ZN9QListData7disposeEPNS_4DataE'
C:/Users/kreij/scoop/apps/msys2/2022-01-28/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: build/release/WbAbstractCamera.o:WbAbstractCamera.cpp:(.text+0xfdd): undefined reference to `__imp__ZN10QArrayData10deallocateEPS_yy'
C:/Users/kreij/scoop/apps/msys2/2022-01-28/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: build/release/WbAbstractCamera.o:WbAbstractCamera.cpp:(.text+0x121d): undefined reference to `__imp__ZN10QArrayData10deallocateEPS_yy'
C:/Users/kreij/scoop/apps/msys2/2022-01-28/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: build/release/WbAbstractCamera.o:WbAbstractCamera.cpp:(.text+0x1256):  [... cut ...]
C:/Users/kreij/scoop/apps/msys2/2022-01-28/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: build/release/WbAccelerometer.moc.o:WbAccelerometer.moc.cpp:(.rdata$_ZTV15WbAccelerometer[_ZTV15WbAccelerometer]+0x430): undefined reference to `non-virtual thunk to WbAccelerometer::~WbAccelerometer()'
collect2.exe: error: ld returned 1 exit status
make[1]: *** [Makefile:650: ../../msys64/mingw64/bin/webots-bin.exe] Error 1
make: *** [Makefile:108: webots_target] Error 2
```

stefaniapedrazzi, 2022-07-14:
Could you try to run `make cleanse` before `make` after switching from `master` to `develop` branch?

There is nothing different in the current `master` and `develop` branch related to Qt and even in the CI on clean Windows machines the exact same workflow works for compiling Webots.
So this issue seems related to your environment.

Kreijstal, 2022-07-14:
Yes, you are right, I thought make would simply make it again without needing it to clean first.

Now I have the more easily solvable Error:
```
user_commands/WbUndoStack.hpp:22:10: fatal error: QtGui/QUndoStack: No such file or directory
   22 | #include <QtGui/QUndoStack>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [Makefile:765: WbActionManager.o] Error 1
make: *** [Makefile:108: webots_target] Error 2
```

omichel, 2022-07-15:
You should uninstall Qt5 and install Qt6 using [scripts/install/qt_windows_installer.sh](https://github.com/cyberbotics/webots/blob/master/scripts/install/qt_windows_installer.sh).

closed 2022-08-29 by BenjaminDeleze