cyberbotics/webots #4906: Wrong "Bounding Object Based" MoI

ShuffleWire, 2022-07-18:
**Describe the Bug**
Using a "Bounding object based" MoI give values that are different from the default MoI values.

<details>
<summary>World</summary>
<pre>
#VRML_SIM R2022a utf8
WorldInfo {
  lineScale 10
}
Viewpoint {
  orientation 0.18483272608525197 0.34484097093526966 -0.9202834172863895 1.2065546190350083
  position -8.08416062735125 16.7740818030343 8.768359541015466
}
TexturedBackground {
}
TexturedBackgroundLight {
}
Robot {
  children [
    DEF s Shape {
      appearance PBRAppearance {
      }
      geometry Sphere {
      }
    }
  ]
  boundingObject USE s
  physics Physics {
    density -1
    mass 1
  }
}
</pre>
</details>

**Steps to Reproduce**
1. Open the world, note that in the "Mass" tab of the Solid the MoI is [0.4,0.4,0.4, 0,0,0] 
That is correct
2. RIght click on "physic/inertiaMatrix" -> Add New -> Bounding Object Based -> Add
3. Observe that the MoI in the "Mass" tab of the Solid change to [1675.52,1675.52,1675.52, 0,0,0]
That change should not occurs, the object haven't changed. This is because the inserted matrix is [1675.52,1675.52,1675.52, 0,0,0]

**Expected behavior**
The MoI inserted into the field inertiaMatrix should be [0.4,0.4,0.4, 0,0,0], to stay consistent with the MoI initially displayed in the "Mass" tab

**System**
 - Operating System: Debian 11
 - Webots: 2022b

Hope I could help in finding a bug :)

stefaniapedrazzi, 2022-07-18:
This seems the exact same issue as reported in  #4684.

ShuffleWire, 2022-07-18:
I don't think so :  
In #4684 , I explaining a bug that is only visible until reload.
Here, the bug is when **generating** a MoI based on the current Bounding Object using the tool discussed in the OP.
Whenever you reload or not don't impact the MoI (being actually written in the wrl file)

ShuffleWire, 2022-07-18:
Additionally, changing the mass of the Physic don't affect the inertiaMatrix, even if the code suggest that it should change.
`
void WbPhysics::updateMass()
  ...
  else if (mMode == BOUNDING_OBJECT_BASED)
    emit massOrDensityChanged();
`
But on that part, I will not be as certain...

ShuffleWire, 2023-06-20:
It have been tested on Webots 2023a-rev1 and it works, thanks you
I'm closing it.

closed 2023-06-20 by ShuffleWire