cyberbotics/webots #4932: Crash when deleting an IndexeFaceSet with a used DEF

ShuffleWire, 2022-07-20:
**Describe the Bug**
On some occasion deleting a DEF node used by another node crash Webots. (I got the warning popup, but sometimes I really want to delete it)

<details>
<summary>World</summary>
<pre>
#VRML_SIM R2022a utf8
WorldInfo {
  lineScale 10
}
Viewpoint {
  orientation -0.43330395888912 0.4393239860899338 0.7869193824382948 1.698763189829365
  position 10.923513114174908 -33.47173612958207 53.27949003590689
}
TexturedBackground {
}
TexturedBackgroundLight {
}
Solid {
  children [
    Transform {
      translation -1.22474 -1.22474 -1.22474
      children [
        Shape {
          appearance PBRAppearance {
          }
          geometry DEF Cube_3645_geometry IndexedFaceSet {
            coord Coordinate {
              point [
                0 0 0
                0 0 2.449
                0 2.449 0
                0 2.449 2.449
                2.449 0 2.449
                2.449 0 0
                2.449 2.449 0
                2.449 2.449 2.449
              ]
            }
            coordIndex [
              0, 1, 2, -1, 0, 2, 6, -1, 0, 4, 1, -1, 0, 5, 4, -1, 0, 6, 5, -1, 1, 3, 2, -1, 1, 4, 7, -1, 1, 7, 3, -1, 2, 3, 7, -1, 2, 7, 6, -1, 4, 5, 6, -1, 4, 6, 7, -1
            ]
          }
        }
      ]
    }
  ]
  boundingObject Group {
    children [
      Transform {
        translation -1.22474 -1.22474 -1.22474
        children [
          Shape {
            geometry USE Cube_3645_geometry
          }
        ]
      }
    ]
  }
  physics Physics {
    density -1
    mass 1
  }
}
</pre>
</details>

**Steps to Reproduce**
1. Open the world
2. Duplicate the solid directly under it
[... 81 lines cut ...]
#51 0x00007ffff7aa9f30 in void doActivate<false>(QObject*, int, void**) () at /home/yann/webots/lib/webots/libQt5Core.so.5
#52 0x00007ffff6160672 in QAction::triggered(bool) () at /home/yann/webots/lib/webots/libQt5Widgets.so.5
#53 0x00007ffff6162a9e in QAction::activate(QAction::ActionEvent) () at /home/yann/webots/lib/webots/libQt5Widgets.so.5
#54 0x00007ffff616339d in QAction::event(QEvent*) () at /home/yann/webots/lib/webots/libQt5Widgets.so.5
--Type <RET> for more, q to quit, c to continue without paging--
#55 0x00007ffff616643c in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /home/yann/webots/lib/webots/libQt5Widgets.so.5
#56 0x00007ffff616cf20 in QApplication::notify(QObject*, QEvent*) () at /home/yann/webots/lib/webots/libQt5Widgets.so.5
#57 0x00007ffff7a71808 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /home/yann/webots/lib/webots/libQt5Core.so.5
#58 0x00007ffff6c92d66 in QShortcutMap::dispatchEvent(QKeyEvent*) () at /home/yann/webots/lib/webots/libQt5Gui.so.5
#59 0x00007ffff6c937e9 in QShortcutMap::tryShortcut(QKeyEvent*) () at /home/yann/webots/lib/webots/libQt5Gui.so.5
#60 0x00007ffff6c453a2 in QWindowSystemInterface::handleShortcutEvent(QWindow*, unsigned long, int, QFlags<Qt::KeyboardModifier>, unsigned int, unsigned int, unsigned int, QString const&, bool, unsigned short) () at /home/yann/webots/lib/webots/libQt5Gui.so.5
#61 0x00007ffff6c60e64 in QGuiApplicationPrivate::processKeyEvent(QWindowSystemInterfacePrivate::KeyEvent*) () at /home/yann/webots/lib/webots/libQt5Gui.so.5
#62 0x00007ffff6c65935 in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) () at /home/yann/webots/lib/webots/libQt5Gui.so.5
#63 0x00007ffff6c418ab in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /home/yann/webots/lib/webots/libQt5Gui.so.5
#64 0x00007fffe718069a in xcbSourceDispatch(_GSource*, int (*)(void*), void*) () at /home/yann/webots/lib/webots/libQt5XcbQpa.so.5
#65 0x00007ffff19e3e6b in g_main_context_dispatch () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#66 0x00007ffff19e4118 in  () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#67 0x00007ffff19e41cf in g_main_context_iteration () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#68 0x00007ffff7acd1cc in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /home/yann/webots/lib/webots/libQt5Core.so.5
#69 0x00007ffff7a7021a in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /home/yann/webots/lib/webots/libQt5Core.so.5
#70 0x00007ffff7a791d3 in QCoreApplication::exec() () at /home/yann/webots/lib/webots/libQt5Core.so.5
#71 0x0000555555a04321 in WbGuiApplication::exec() (this=0x7fffffffdd50) at gui/WbGuiApplication.cpp:343
#72 0x0000555555c2418b in main(int, char**) (argc=1, argv=0x7fffffffdfa8) at gui/main.cpp:207
</pre>
</details>

BenjaminDeleze, 2022-07-22:
I can reproduce the crash on Ubuntu22, master.

From what I have investigate:
- It happens only if after a copy (if copy->save->reload: no crash).
- Cut-paste does not make it crash.
- The structure group->transform->shape->geometry used in the BO is mandatory to make it crash.
- It seems that when it crashes, there is an recursion between `WbMatter::updateOdePlaceableGeomPosition` and `WbMatter::updateOdeGeomPosition` with a growing n at https://github.com/cyberbotics/webots/blob/26cd4e59879785b6c17d211b443b0212da9c5bef/src/webots/nodes/WbMatter.cpp#L614
- If I comment this line https://github.com/cyberbotics/webots/blob/26cd4e59879785b6c17d211b443b0212da9c5bef/src/webots/nodes/WbMatter.cpp#L597, it stops the infinite recursion.

From what I can see, I am tempted to say that there is a dangling pointer somewhere or a memory leak. It would explain that we need very specific conditions to make the crash reproducible.

BenjaminDeleze, 2022-07-29:
I think I managed to isolate the infinite loop with GDB (from bottom to top):

```
QObject::~QObject() () at /home/benjamin/webots2/lib/webots/libQt6Core.so.6
WbMatter::updateOdePlaceableGeomPosition(dxGeom*) (this=0x55555c7bff50, g=0x555556be3a20) at nodes/WbMatter.cpp:591
WbMatter::updateOdeGeomPosition(dxGeom*) (this=0x55555c7bff50, g=0x55555c7f2ed0) at nodes/WbMatter.cpp:626
WbMatter::updateOdeGeomPosition() (this=0x55555c7bff50) at nodes/WbMatter.hpp:81
WbSolidMerger::setGeomOffsetPositions() (this=0x55555c7c17a0) at nodes/utils/WbSolidMerger.cpp:168
WbSolidMerger::setGeomAndBodyPositions(bool, bool) (this=0x55555c7c17a0, zeroVelocities=0x0, resetJoints=0x1) at nodes/utils/WbSolidMerger.cpp:359
WbSolidMerger::mergeMass(WbSolid*, bool) (this=0x55555c7c17a0, solid=0x55555c7bff50, subtract=0x1) at nodes/utils/WbSolidMerger.cpp:247
WbSolid::adjustOdeMass(bool) (this=0x55555c7bff50, mergeMass=0x1) at nodes/WbSolid.cpp:990
WbSolid::addMassFromInsertedNode(WbBaseNode*) (this=0x55555c7bff50, node=0x55555c134610) at nodes/WbSolid.cpp:1001
WbSolid::setGeomMatter(dxGeom*, WbBaseNode*) (this=0x55555c7bff50, g=0x555556ac7b80, node=0x55555c134610) at nodes/WbSolid.cpp:1197
WbMatter::createOdeGeomFromInsertedTransformItem() (this=0x55555c7bff50) at nodes/WbMatter.cpp:359
QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, void (WbMatter::*)()>::call(void (WbMatter::*)(), WbMatter*, void**) (f=(void (WbMatter::*)(WbMatter * const)) 0x555555d0eea0 <WbMatter::createOdeGeomFromInsertedTransformItem()>, o=0x55555c7bff50, arg=0x7fffffffbea0) at ../../include/qt/QtCore/QtCore/qobjectdefs_impl.h:152
QtPrivate::FunctionPointer<void (WbMatter::*)()>::call<QtPrivate::List<>, void>(void (WbMatter::*)(), WbMatter*, void**) (f=(void (WbMatter::*)(WbMatter * const)) 0x555555d0eea0 <WbMatter::createOdeGeomFromInsertedTransformItem()>, o=0x55555c7bff50, arg=0x7fffffffbea0) at ../../include/qt/QtCore/QtCore/qobjectdefs_impl.h:185
QtPrivate::QSlotObject<void (WbMatter::*)(), QtPrivate::List<>, void>::impl(int, QtPrivate::QSlotObjectBase*, QObject*, void**, bool*) (which=0x1, this_=0x55555c7f2fd0, r=0x55555c7bff50, a=0x7fffffffbea0, ret=0x0) at ../../include/qt/QtCore/QtCore/qobjectdefs_impl.h:395
() at /home/benjamin/webots2/lib/webots/libQt6Core.so.6
WbTransform::geometryInTransformInserted() (this=0x55555c134610) at build/debug/WbTransform.moc.cpp:213
WbTransform::createOdeGeom(int) (this=0x55555c134610, index=0xffffffff) at nodes/WbTransform.cpp:281
WbTransform::createOdeGeomIfNeeded() (this=0x55555c134610) at nodes/WbTransform.cpp:257
QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, void (WbTransform::*)()>::call(void (WbTransform::*)(), WbTransform*, void**) (f=(void (WbTransform::*)(WbTransform * const)) 0x555555f1b7dc <WbTransform::createOdeGeomIfNeeded()>, o=0x55555c134610, arg=0x7fffffffc150) at ../../include/qt/QtCore/QtCore/qobjectdefs_impl.h:152
QtPrivate::FunctionPointer<void (WbTransform::*)()>::call<QtPrivate::List<>, void>(void (WbTransform::*)(), WbTransform*, void**) (f=(void (WbTransform::*)(WbTransform * const)) 0x555555f1b7dc <WbTransform::createOdeGeomIfNeeded()>, o=0x55555c134610, arg=0x7fffffffc150) at ../../include/qt/QtCore/QtCore/qobjectdefs_impl.h:185
QtPrivate::QSlotObject<void (WbTransform::*)(), QtPrivate::List<>, void>::impl(int, QtPrivate::QSlotObjectBase*, QObject*, void**, bool*) (which=0x1, this_=0x55555c7f24a0, r=0x55555c134610, a=0x7fffffffc150, ret=0x0) at ../../include/qt/QtCore/QtCore/qobjectdefs_impl.h:395
() at /home/benjamin/webots2/lib/webots/libQt6Core.so.6
QObject::destroyed(QObject*) () at /home/benjamin/webots2/lib/webots/libQt6Core.so.6
QObject::~QObject() () at /home/benjamin/webots2/lib/webots/libQt6Core.so.6
```

PR #5686 (2022-12-23): Fix infinite loop updating ODE geometries
Fix #4932:
trying to update the ODE position of a geometry that is being deleted causes the emission of a `WbTransform::destroyed` signal that triggers the generation of new ODE geometries in the same solid merger instance and a new update of all the geometries.
This results in an infinite loop with increasing number of geometries that is never correctly completed.

This behavior occurs because in `WbGeometry` it can happen that the link to the ODE geom object is overwritten without destroying the previous ODE geom.
Thus, in particular when updating th [... cut]
closed 2022-12-23 by omichel