cyberbotics/webots #4946: Webots window blocked and black at start

stefaniapedrazzi, 2022-07-26:
When lot of `IMPORTABLE EXTERNPROTO` are declared in a file Webots takes very long to start and load a world.
While loading, the Webots window is completely black (if it is open at start), it is blocked and there is no dialog message showing the loading progress with the option to cancel the load that it is very handy for big simulations.

Steps to reproduce:
1. open the `scripts/icon_studio/worlds/icon_studio.wbt`
-> after a while I even gen the message that `Webots` is not responding and I'm asked if I want to "Force Quit" and as I user I would think that Webots is buggy and it freezed

I tested on Ubuntu 22.04.

Loading all the models takes long, but we should at least show and update the loading progress bar and add the option to cancel the load.

stefaniapedrazzi, 2022-07-26:
This also occurs with other complex worlds but takes less time (for example the `mavic_2_pro.wbt`).
1. Loading window shows text `Parsing
2. Loading window shows text  Downloading assets`
3. Loading window becomes black and unresponsive
4. Loading window shows text  `Finalizing nodes`

omichel, 2022-07-26:
I believe we should not load the importable externproto models until they are actually imported. That would fix the issue and also be better in particular with a use case where you set a large number of importable externproto but use randomly only a couple of them during a simulation run.

ad-daniel, 2022-07-26:
What do you intend by "loading"? IMPORTABLE EXTERNPROTO nodes aren't created until they are spawned, they are on the other hand downloaded when parsing the world, which was the intended behavior since we don't want to halt the simulation, wait for things to download, and proceed only then to the next timestep. Moreover, PROTO flagged as "IMPORTABLE EXTERNPROTO" can also be instantiated at load-time (i.e., IMPORTABLE EXTERNPROTO supersedes simple EXTERNPROTO to avoid having 2 separate declarations), as such a-priori we cannot know which among the IMPORTABLE EXTERNPROTO we'll need "only later".

Icon studio has ~300+ PROTO at level 0, which in turn depend in many more sub-proto and each of those depends on tons of textures. Although when run locally there's no "download" per-se taking place (since everything is local), we still need to open each of them, apply several regex to extract the declarations and so on, which I assume is what is taking so long.

Also note there's 2 stages of "Download assets", one is for the PROTO (taking place before the loading of the world), one is for the other assets (textures, sounds, ..), which is done during the loading. In-between the two I assume it's creating the nodes.

omichel, 2022-07-26:
If the overhead comes from the regex parsing of proto files, a good solution would be to clean-up proto files by replacing all the large `IndexedFaceSet` nodes with `.obj` files.

ad-daniel, 2022-07-26:
The PROTO of the mavic is 5.6MB, the default ubuntu 20 text editor struggles to even open it, applying regex to such a file might very well be the cause of what you mention

omichel, 2022-07-26:
The [large_protos.txt](https://github.com/cyberbotics/webots/files/9188008/large_protos.txt) file lists all the proto files by size. We actually have many large protos, including the `Mavic2Pro`. Getting rid of the large `IndexedFaceSet` nodes in there would certainly improve the situation. Probably all the protos could be reduced to a size inferior to 8 or 10KB.

stefaniapedrazzi, 2022-07-28:
If nobody is looking at this issue I can start reducing the size of the Mavic2Pro and check if the issue is solved.

stefaniapedrazzi, 2022-08-02:
Reducing the size of the Mavic2Pro file doesn't seem to solve the issue.
During the load of the world, the loading window still becomes completely black and unresponsive but it should not.
The purpose of the loading window is exactly to tell users that Webots is doing something and it is not blocked, so it is very bad it is not regularly updated during the load.
I will inspect more deeply where is the issue.

stefaniapedrazzi, 2022-08-02:
The load is very slow due to the continuous call of the `WbExternProtoEditor::updateContents` method.
I'm fixing it and I will open a PR very soon.

stefaniapedrazzi, 2022-08-02:
https://github.com/cyberbotics/webots/pull/4992 improved the loading speed.
However, the issue with the loading window is not solved (I'm working on it, but I won't be able to complete this week).

closed 2022-08-03 by ad-daniel