cyberbotics/webots #4952: EXTERNPROTO declaration is parsed even if added in comment

stefaniapedrazzi, 2022-07-26:
It seems that even if the EXTERNPROTO declaration is added in a comment, Webots parses and processes it.

Here is a sample world file:
```
#VRML_SIM R2022b utf8
# EXTERNPROTO "../protos/Ball.proto"

EXTERNPROTO "https://raw.githubusercontent.com/cyberbotics/webots/R2022b/projects/objects/floors/protos/RectangleArena.proto"

WorldInfo {
}
Viewpoint {
}
RectangleArena {
}
```

Error in the console:
```
ERROR: '/home/stefi/Desktop/test_project1/worlds/test2.wbt': Failed to load due to invalid token(s).
```

ygoumaz, 2022-07-27:
I have investigated the problem and it seems that the EXTERNPROTOs in comments are not processed, which is correct. The error comes from the fact that there is a comment directly after the first line `#VRML_SIM R2022b utf8` without any space. I believe a space is required here for the header to be valid and the PROTO to be parsed correctly.

stefaniapedrazzi, 2022-07-27:
I will check again.
Because Webots was crashing even if I was commenting the EXTERNPROTO causing the issue.

stefaniapedrazzi, 2022-07-27:
I confirm the issue is just the space.

However, even in this case Webots should throw an error and load the default world instead of the current behavior:
on the VM it freezes, on my development system it seems to partially load something but crashes as soon as I click on the step button.

stefaniapedrazzi, 2022-07-27:
Additionally there is no reason why a new line is needed between the header and the comment.
Also note that this is valid in VRML.

omichel, 2022-07-29:
I assume we can now close this issue.

closed 2022-07-29 by omichel