cyberbotics/webots #4953: Webots freezing with absolute PROTO path and missing warning for PROTO with same name

stefaniapedrazzi, 2022-07-26:
1. I get not warning in the console when using different URLs for a PROTO with the same name in the world file and in the PROTO file.
For example in the world file I have this declaration:
    ```
    EXTERNPROTO "/home/stefi/Desktop/Ball.proto"
   ```
    and in the PROTO file I have this that points to "/home/stefi/Desktop/test_project1/protos/Ball.proto"
    ```
    EXTERNPROTO "../protos/Ball.proto"
    ```
    I get no warnings and the one defined in the world file is used.

2. If I replace the declaration in the world file and use the absolute local path:
    ```
    EXTERNPROTO "/home/stefi/Desktop/Ball.proto"
    ```
    Webots freezes when loading the world.
    This PROTO cannot be added from the UI, but Webots should handöle absolute paths.

omichel, 2022-07-26:
1. I believe it is normal that no warning is displayed if EXTERNPROTO declarations for the same PROTO name are defined both in the world file and in a PROTO file. If the PROTO is instantiated in the world file, the declaration of the world file should be used. If the PROTO is instantiated inside another PROTO, the declaration in the containing PROTO file should be used. Does your case match this?

stefaniapedrazzi, 2022-07-26:
N, it doesn't work because currently Webots supports only a single source for a PROTO for the whole simulation.
So the declaration in the world is used also for the PROTO.

omichel, 2022-07-26:
> No, it doesn't work because currently Webots supports only a single source for a PROTO for the whole simulation. So the declaration in the world is used also for the PROTO.

Unfortunately, this is wrong.

stefaniapedrazzi, 2022-07-26:
I will further inspect the crash (issue 2.)

stefaniapedrazzi, 2022-07-26:
I addressed issue 2 in https://github.com/cyberbotics/webots/pull/4957.
I will let @ad-daniel address issue 1.

omichel, 2022-08-05:
@ad-daniel: I believe we can close this issue now.

ad-daniel, 2022-08-05:
Indeed

closed 2022-08-05 by ad-daniel