cyberbotics/webots #4962: Web simulation quick start not working

k-berryman, 2022-07-26:
The [web simulation instructions](https://www.cyberbotics.com/doc/guide/web-simulation#quick-start) quick start instructions seem to be broken.

These steps work fine: 
`sudo apt install subversion firejail python3-tornado python3-pynvml`
`cd $WEBOTS_HOME/resources/web/server`
`./server.sh start default`
and I can open `$WEBOTS_HOME/resources/web/streaming_viewer/index.html` in my browser successfully

[image]

---

This step is where the errors start:

> In the user interface, find a Connect to field, and type for example:
`ws://localhost:1999/session?url=webots://github.com/cyberbotics/webots/branch/develop/projects/languages/python/worlds/example.wbt`

I've tried variations of that URL, but with no luck.

---

**Expected Behavior**
The expected behavior looks like [this example](https://cyberbotics1.epfl.ch/open-roberta/)

---

**Screenshots**
Here are a few screenshots of my errors

[image]

[image]

---

**System**
Webots version: R2022a
OS: Ubuntu 20.04.4 LTS
Branch: I have no local repo for Webots

**Additional context**
I made a post in Discord & Benjamin Délèze asked for me to make this issue.

BenjaminDeleze, 2022-07-27:
Thank you, for reporting the issue.

It seems that it is a permission problem. 
From what I could see, I think that you have installed Webots with the `.deb`. Right?

To solve the issue:

- Copy the server folder from `/usr/local/webots/resources/server` to `/home/YourName/Documents`.
- Open the `default.json` file located in `/home/YourName/Documents/server/config/simulation/` with you ide/text editor and add the following line between the `{}`: `"webotsHome": "/usr/local/webots"`. This line is used to tell the server where Webots is located
- Concerning the url, you should use this one: `ws://localhost:1999/session?url=webots://github.com/cyberbotics/webots/tag/R2022a/projects/languages/python/worlds/example.wbt`. The one in the documentation does not work because some new features that are on the `develop` branch are not compatible with `R2022a`

k-berryman, 2022-07-27:
Yes, I installed `webots_2022a_amd64.deb`. My Downloads folder shows that it has these permissions `-rw-rw-r--`.

Your solution works for me! Thank you so much.

I specified my steps below anyways.
- I copied to the server folder to my Documents folder using `cp -R /usr/local/webots/resources/web/server/ ~/Documents/`
- Edited the contents of `default.json` as suggested
- In `~/Documents/server`, `./server.sh start default` which works correctly and prints the following
> Started default session server (6638)
> Started default simulation server (6639)
- Went to `/usr/local/webots/resources/web/streaming_viewer/index.html` in Chrome
- Used the new url

Thank you again

BenjaminDeleze, 2022-07-28:
Thank you for your feedback, I will update the instructions

closed 2022-07-28 by BenjaminDeleze