cyberbotics/webots #4964: Confusion between PROTOs that should be distinguished

ygoumaz, 2022-07-27:
If 2 PROTOs (1 local and 1 remote) have the same name, adding one of them to the list of IMPORTABLE EXTERNPROTOs will always spawn the remote one.

**Steps to Reproduce**
1. Create a new PROTO with the PROTO wizard from any object (e.g. Cabinet). Give it the same name as an official Webots PROTO, but different than the PROTO you have created it from (e.g. Nao).
2. Open the IMPORTABLE EXTERNPROTO list editor.
3. Click on `Insert new`.
4. Select the local PROTO you have created in "current project".
5. Spawn this PROTO with a supervisor.
6. The official Nao will be imported, instead of the Cabinet named Nao.

Moreover, after step 4, if the world is saved, the Nao is declared in the world file as IMPORTABLE EXTERNPROTO but with the remote URL and not the local one.

This behavior is probably 'normal' right now, as we don't allow multiple PROTOs with same names. But #4953

omichel, 2022-08-04:
Can we close this one, as it seems it was fixed in #4967?

ad-daniel, 2022-08-04:
Indeed

closed 2022-08-04 by ad-daniel