cyberbotics/webots #5024: Scaling issue in simulation UI R2022b

TheMangalex, 2022-08-08:
**Describe the Bug**
I hope this isn't a duplication, but I didn't find a fix or something related. It might also be an issue with 4k displays.
Since the update to R2022b, the simulation window fills only a quarter of the given screen space. 

The mouse movements and clicks are measured at the whole screen though, making it additionally hard to click on objects.

**Steps to Reproduce**
1. Installing webots from source origin/master (make -j8)
2. Launching with ./webots

**Expected behavior**
Normal screen scaling as seen in 2nd Screenshot

**Screenshots**
R2022b
[image]

R2022a
[image]

**System**
 - Operating System: Ubuntu 20.04
 - Graphics Card: Intel HD Graphics & Rtx 3070 (only tested on intel currently though)
 - Display Settings: 4k (3840x2160)
 - Window-Manager: i3

**Additional context**
Add any other context about the problem here.

omichel, 2022-08-08:
That's is strange... Can you try to to set `export QT_USE_PHYSICAL_DPI=1` and run Webots again?
Alternatively, can you also try `export QT_SCALE_FACTOR=2`?
Also can you try to remove `QT_ENABLE_HIGHDPI_SCALING` from https://github.com/cyberbotics/webots/blob/master/src/webots/launcher/webots-linux.sh#L74?

Let me know your findings.

TheMangalex, 2022-08-08:
Okay so the 1st and 3rd option didn't change anything, but the 2nd option looked like that
[image]
.
Also it seems like this bug doesn't happen in Gnome for me and only i3, but i3 was working correctly with R2022a. Maybe some variable is incorrectly provided by the X-Server?

omichel, 2022-08-08:
Maybe [this one](https://github.com/cyberbotics/webots/commit/88e8b517af05661aa55d7e7e2209ba12e04bb894) broke it? Can you try to revert it?

TheMangalex, 2022-08-08:
Sadly this doesn't seem to change it.

TheMangalex, 2022-08-08:
Sadly nothing as well.

omichel, 2022-08-08:
Unfortunately, I am running out of ideas... That's strange the bug doesn't show up with R2022a...
A big change we did between R2022a and R2022b is the [upgrade to Qt6](https://github.com/cyberbotics/webots/pull/4189).
To find the problem, we could revert back in the github history to some commit before the upgrade to Qt6 and rebuild Webots to try to identify if problem was introduced here...

TheMangalex, 2022-08-08:
Thank you very much for your help. Maybe there is a general change in how qt6 handles scaling and that makes problems with my display config. I'll check it and if I find a solution, I'll post an update.

TheMangalex, 2022-08-09:
Interestingly using a QT_SCALE_FACTOR of 0.5 fixes the screen scaling, although the rest of the ui is pretty small.
Is there a way to increase the ui size (other than with QT_SCALE_FACTOR)?
[image]

omichel, 2022-08-09:
We are progressing... Did you check all the different environment variables documented here: https://doc.qt.io/qt-6/highdpi.html#environment-variable-reference?

omichel, 2022-08-09:
Alternatively, you may want to try to change the font sizes in the styling sheet: https://github.com/cyberbotics/webots/blob/master/resources/stylesheet.linux.qss

TheMangalex, 2022-08-09:
Explicitely disabling the QT_ENABLE_HIGHDPI_SCALING with QT_ENABLE_HIGHDPI_SCALING=0 makes the Ui small, but working. I somehow forgot to check disabling it yesterday and tried just not to enable it.
In combination with a SCALE_FACTOR of 2 and a FONT_DPI of 80 it looks good again.

Thank you very much for your help!
[image]

omichel, 2022-08-09:
Could we integrate this in https://github.com/cyberbotics/webots/blob/master/src/webots/launcher/webots-linux.sh?
If we can detect that we are in a high DPI i3wm environment, the launcher script could set these environment variables automatically?

TheMangalex, 2022-08-09:
That sounds like a good option.
I used the parameters

export QT_ENABLE_HIGHDPI_SCALING=0
export QT_SCALE_FACTOR=2
export QT_FONT_DPI=80

to fix my scaling. The Environment variable  $XDG_CURRENT_DESKTOP returns i3 for me.
I might be able to adapt the script later today and test it.

omichel, 2022-08-09:
That would be great.

closed 2022-08-23 by TheMangalex