cyberbotics/webots #5058: Fail to load extern Proto from another github repo

BenjaminDeleze, 2022-08-11:
Webots crash with the following world:

```
#VRML_SIM R2022b utf8

EXTERNPROTO "https://raw.githubusercontent.com/cyberbotics/webots-cloud-simulation-examples/test-protos-download/7_local_assets/protos/MyProto.proto"

WorldInfo {
}
Viewpoint {
  orientation -0.0316302248353415 0.02843866445440257 0.9990949760863156 1.6779495280745207
  position -0.05753688709726134 2.5600691798403346 9.999992923742619
}
Background {
  skyColor 0.5 0.5 0.8
}

MyProto {
}

```

`MyProto` is a valid Proto and works well if you open this world: https://github.com/cyberbotics/webots-cloud-simulation-examples/blob/main/7_local_assets/worlds/local_assets_example.wbt

ad-daniel, 2022-08-11:
No crashes for me in development mode
[image]

BenjaminDeleze, 2022-08-11:
I think it is a problem with CadShape, can you try with `EXTERNPROTO "https://raw.githubusercontent.com/cyberbotics/webots-cloud-simulation-examples/main/7_local_assets/protos/MyProto.proto"` instead?

If it is the case, #5015 will maybe fix it

ad-daniel, 2022-08-11:
Yes it crashes, the issue is in WbUrl, it triggers an assert when trying to resolve the obj relative to the PROTO

ad-daniel, 2022-08-11:
I have an idea why it would fail, the ImageTextures used by CadShape are shallow nodes... We might need to unshallow them a bit and define a parent regardless

BenjaminDeleze, 2022-08-15:
It seems that it has been fixed by #5015, I can now load remote CadShape correctly

closed 2022-08-17 by ad-daniel