cyberbotics/webots #5066: Remote PROTO without icon throw download warning

ad-daniel, 2022-08-11:
**Describe the Bug**

Take this example of world, the PROTO is downloaded from a remote url outside our own. When selecting this PROTO in the add-node window, a warning is shown saying that the icon could not be downloaded (none exists in the repo). By printing the error message we're essentially forcing people to define an icon for their PROTO which isn't ideal. If it couldn't be downloaded we should either show nothing (as we do for local proto) or show a default one, without printing any warning.

```
#VRML_SIM R2022b utf8

EXTERNPROTO "https://raw.githubusercontent.com/cyberbotics/webots-cloud-simulation-examples/main/7_local_assets/protos/MyProto.proto"

WorldInfo {
}
Viewpoint {
  orientation -0.0316302248353415 0.02843866445440257 0.9990949760863156 1.6779495280745207
  position -0.05753688709726134 2.5600691798403346 9.999992923742619
}
Background {
  skyColor 0.5 0.5 0.8
}
MyProto {
}
```

BenjaminDeleze, 2022-08-11:
<p>
[image]
[image]
[image]
[image]
</p>

What do you think of those default icons?

omichel, 2022-08-11:
I prefer the "Missing PROTO icon" (more similar to "Missing texture"), but I would capitalize "PROTO".

BenjaminDeleze, 2022-08-11:
After some discussion we decided to go with the following version to match the `MISSING TEXTURE` image:
<p>
[image]
[image]
</p>

PR #5068 (2022-08-11): Fix warning missing icon
Fix #5066
closed 2022-08-11 by BenjaminDeleze