cyberbotics/webots #5072: URL relative to an unsaved world

ad-daniel, 2022-08-12:
**Describe the Bug**

Based on the visibility and/or while streaming, some URL may be expressed or searched for relative to the world file. This however presupposes that the world itself has a known location. If a user creates a world, adds stuff but never actually saves, the behavior is incorrect.

omichel, 2022-08-15:
That is an edge use case. Possible solutions include:
1. Enforcing a world to be saved before streaming.
2. Enforce saving the world when creating it.

I believe solution 2 is the best as it seems needed to know the location of the world as soon as we want to modify it and add some objects with textures, meshes, etc.

omichel, 2022-08-15:
To achieve this, I suggest to simply remove the "File / New World" which seems to be redundant and inconsistent with the Wizard menu.

BenjaminDeleze, 2022-08-15:
> To achieve this, I suggest to simply remove the "File / New World" which seems to be redundant and inconsistent with the Wizard menu.

I am not sure to agree as sometime you just want to add a world to an existing project

PR #5075 (2022-08-18): Move Wizard menu in File/New menu
Fixes  #5072.

This PR moves the Wizard menu into the File / New menu as it is a more natural place to create new projects, worlds, controllers, plugins, protos, etc. It also fixes #5072 by enforcing the naming and saving of the world file when creating a new world.
closed 2022-08-18 by omichel