cyberbotics/webots #5076: Broken textures

BenjaminDeleze, 2022-08-15:
Some textures are currently broken on master, for example in `highway_overtake` and `village_realistic`.

It seems that some textures where broken by https://github.com/cyberbotics/webots/pull/5003
And some other by https://github.com/cyberbotics/webots/pull/5071

BenjaminDeleze, 2022-08-16:
I just check a bit the first part of the issue:
Since #5003, webots does not found anymore the textures for the left and right indicators of the `lincoln` car like `textures/lincoln_mkz_details_base_color.jpg`

It is because, webots try to retrieve those textures from the `vehicles/protos/abstract/textures/` folder instead of the `vehicles/protos/lincoln` one.
This could be partially explained because the indicators are parameters of the the `car` proto which is in the `abstract` folder. However it is somehow more specific because the `textures/lincoln_mkz_details_base_color.jpg` image is used at another place in the `lincoln` proto and is retrieved correctly in that case.

ygoumaz, 2022-08-16:
The issue is probably mainly coming from #5071. The implemented condition in `WbUrl::computePath` seems wrong when dealing with derived PROTOs. The alias value is set to the lowest field, which is (in this case) "url". When looking in the first ancestry of the list, if the "url" field is not directly exposed, the first ancestry will always be chosen. This way the `TiagoIron` works because `TiagoBase` should be chosen. But in the case of the `ToyotaPriusWheel`, the `VehicleWheel` should not be chosen but it actually is.

Two things should probably be adapted:
- The alias to look for. Currently, the "url" field is not correct, because there is only little chance that it is contained in the parameterAliases list of the protoModel.
- In the case of Slots, there is probably a special case to handle, but not completely sure.

Currently, I don't see any easy solution to solve this problem.

ad-daniel, 2022-08-17:
I think fundamentally it isn't wrong to follow the aliases from the deepest ancestor. Derived PROTO by design build upon the PROTO they depend on, in most cases the fields remain common to all ancestors (and new ones are added on top), the logic handles the case if at different levels the name of the field itself changes, which is also possible although less common.

The issue as you say is that if it's derived, we search among the ancestors starting from the bottom (in order to be able to follow the aliases, if any). In the case of the car-related PROTO, all textures work except the `logo` related ones (for the BMW at least), and this is defined in `BmwX5Wheel` directly, not any of the derived PROTO (`VehicleWheel`). Hence by jumping directly to the oldest ancestor, and stopping immediately because it doesn't expose any "url" field (these appearances are defined in a `logoSlot`), we assume that oldest PROTO is the reference one.

Essentially, the relative path in question is relative to a derived PROTO, but the two scenarios which are equally probable are:
- the ancestor defines the texture internally so no alias chain exists hence we need to stop immediately at the oldest ancestor
- the most recent descendant defines the texture internally, so no need to even explore the ancestors

It's tricky as one wants to search bottom up, and the other top-down and both don't have any PROTO parameter involved (the sudden lack of PROTO parameter is what we use as criterion to determine that the field became internal and that the search must stop), so at first glance I don't see an easy solution either, but I'll think about it

closed 2022-08-31 by ad-daniel