cyberbotics/webots #5078: Wrong Missing declaration message

omichel, 2022-08-17:
When opening this world file:

<details>
<summary>my_world_file.wbt</summary>

```js
#VRML_SIM R2022b utf8

EXTERNPROTO "webots://projects/objects/backgrounds/protos/TexturedBackground.proto"
EXTERNPROTO "webots://projects/objects/backgrounds/protos/TexturedBackgroundLight.proto"
EXTERNPROTO "webots://projects/objects/tables/protos/RoundTable.proto"
EXTERNPROTO "webots://projects/appearances/protos/CorrugatedPlates.proto"

WorldInfo {
}
Viewpoint {
  orientation -0.5773502691896257 0.5773502691896257 0.5773502691896257 2.0944
  position 0 0 10
}
TexturedBackground {
}
TexturedBackgroundLight {
}
RoundTable {
  appearance CorrugatedPlates {
  }
}
```

</details>

I am getting a random error (happens once out of 5 runs):
```
ERROR: Missing declaration for 'TexturedBackground', add: 'EXTERNPROTO "../../../../../msys64/home/Olivier/webots-develop/projects/objects/backgrounds/protos/TexturedBackground.proto"' to 'C:/Users/Olivi/Desktop/my_project/worlds/my_world_file.wbt'.
```

ad-daniel, 2022-08-17:
By run do you mean starting webots from zero every time or when reloading 5 times?

omichel, 2022-08-17:
Starting from zero.

ad-daniel, 2022-08-17:
Normally that message is part of the backwards compatibility mechanism, which should trigger only if the declaration is missing in the world file (which clearly isn't), so either the call to [findExternProtoDeclarationInFile](https://github.com/cyberbotics/webots/blob/ba268b915810a250d5b48c4a3e8bf757679f0a6b/src/webots/vrml/WbProtoManager.cpp#L133) fails or it never has the chance to occur, which would happen if the recently added `mImportedFromSupervisor` is undefined at this point, which would explain why sometimes it works and others it doesn't. Might be a clue.

closed 2022-08-17 by omichel