cyberbotics/webots #5111: Meshes are loaded too many times

omichel, 2022-08-23:
If you create a very simple world with only a `RubberDuck` proto and if you add a qDebug() statement in https://github.com/cyberbotics/webots/blob/b42cbe6c9ae0c009c0b11436843325e8508f1a01/src/webots/nodes/WbMesh.cpp#L162, you will see that each `Mesh` node of the `RubberDuck` proto is created 3 times by calling the assimp importer function. I believe we should fix this so that they are created only once, as parsing an OBJ file is a costly operation.

BenjaminDeleze, 2022-08-24:
I just looked at it out of curiosity and :
In R2022a and R2021b, the `updateTriangleMesh()` function is called twice for each Mesh if we are in a Proto ( in `WbTriangleMeshGeometry::createTriangleMesh()` and in `WbTriangleMeshGeometry::createWrenObjects()`) and only once if we are not (only in `WbTriangleMeshGeometry::createTriangleMesh()`)

It seems that it passes an additionnal time in `WbTriangleMeshGeometry::createTriangleMesh()` since #4238

BenjaminDeleze, 2022-08-24:
Finally I pushed a little bit further and found the reason behind that:
In #4238, the call to `WbMesh::updateUrl()` was moved from `preFinalize()` to `postFinalize()`.

Thus the condition: 
```
if (areWrenObjectsInitialized()) {
    buildWrenMesh(true);
    if (n > 0)
      emit wrenObjectsCreated();  // throw signal to update pickable state
  }
```
become true (and `buildWrenMesh(true)` reload the mesh file)

omichel, 2022-08-24:
Thanks to #5113, meshes are loaded only twice instead of three times.
However, this is not fully satisfactory as should ideally load meshes only once for obvious performance reasons.
Therefore I am re-opening this issue.
I believe we should aim at loading meshes with assimp only once per mesh file.

PR #5113 (2022-08-24): Fix additional mesh loading
Fix #5111 

I just replace `updateUrl()` at its initial place.

I could not find any side effect to that. Either the function was moved by error or the thing it was supposed to fix was also fixed by something else in the meantime.
PR #5117 (2022-08-25): Load meshes only once
Fix #5111 

The meshes were loaded twice when in a Proto:
- once in `WbTriangleMeshGeometry::createTriangleMesh()`
- once in `WbTriangleMeshGeometry::createWrenObjects()`

By removing the call in `createWrenObjects` the mesh are now loaded only once.

However, I could not find/understand why there was this additional call in `createWrenObjects`. All that I know is that it is very old (prior to R2019).

I tested all the functionnalities I thought of and everything seems fine (regeneration, convert to base node, test suite, load mesh in proto,...).

I anybody know more abou [... cut]
closed 2022-08-25 by BenjaminDeleze