cyberbotics/webots #5131: Textures copied outside current project during relocation

stefaniapedrazzi, 2022-08-29:
When relocating a world that is using relative URLs for texture, a corresponding folder is created during project relocation at the same path but relative to the new project location.

This is fundamentally wrong and Webots should never modify or copy anything on the system outside of the location chosen by the user.

**Steps to reproduce**
1. Open sample `track.wbt`
2. Add a texture URL pointing to a relative path to the node `DEF TRACKED_ROBOT Robot > children > DEF GEOM Shape > appearance`: for example "../../../default/worlds/textures/square_block_parquetry.jpg" (path defined by Webots when choosing a default texture from the Select File dialog)
3. Save the world and accept the project relocation: for example at `~/Desktop`
-> the project relocation also creates the folder `~/default/worlds/textures/` containing `square_block_parquetry.jpg

ad-daniel, 2022-08-30:
A simpler approach could be to just re-define the relative URLs with respect to the new world location instead of the old one. My understanding is that only the relative URLs are causing issues, and doing so would avoid both using an absolute URL and unnecessary copies of the files. We already need to open the world file to know if it references any forest, so it would just require changing the relative URLs at the same time. Am I missing something?

omichel, 2022-08-30:
The problem with this approach is that you cannot move the project folder or it will break all the relative URLs going outside of the project.

closed 2022-08-31 by ad-daniel